openhwfoundation/core-v-mcu #20: Consider using verible for linting and code formatting

mithro, 2020-10-20:
Hi!

It looks like this project is using the LowRISC style guide. You thus might be interested in [Google's verible project](https://github.com/google/verible) which provides an Apache 2.0 licensed SystemVerilog linter and code formatter with the LowRISC style guide as an early target!

Due to the permissive license, you can even run the linter on CI systems like Travis CI, GitHub actions or Azure Pipelines to check that your code stays in compliance!

Good luck with your project!

MikeOpenHWGroup, 2021-11-04:
Ouch, I cannot believe we let this one get so stale.

@mithro, AFAIK we have been using Verible for this project since 2021-06 (see our [top-level README](https://github.com/openhwgroup/core-v-mcu#verible-format-check))

@gmartin102 and/or @zarubaf can you confirm?

gmartin102, 2021-11-04:
Yes we are using verible linter as part of the CI scripts

closed 2021-11-04 by gmartin102

mithro, 2021-11-04:
@gmartin102 - You might be interested in the recent GitHub Actions we published @ https://github.com/chipsalliance/verible-linter-action and https://github.com/chipsalliance/verible-formatter-action which are talked about @ https://antmicro.com/blog/2021/08/verible-integration-with-github-actions/ -- They surface the lint and formatting changes in the GitHub Pull Request UI as shown below;
[image]

MikeOpenHWGroup, 2021-11-04:
Thanks @gmartin102!  Thanks @mithro - I'll be sure to that check out.