openhwfoundation/core-v-mcu #33: GDB in prebuilt pulp toolchain lack of tui support

ericsunplus, 2020-12-25:
## Bug Title
GDB in prebuilt pulp toolchain lack to tui support

### Type
* Toolchain

### Description
I used the `riscv32-unknown-elf-gdb` in suggested [Embecosm PULP GCC toolchain](https://www.embecosm.com/resources/tool-chain-downloads/#pulp) to connect to a cv32e40p FPGA. The debugging can work, but it is lacking of tui support so can't achieve interactive debugging

```
$ riscv32-unknown-elf-gdb --tui build/bubble/bubble
riscv32-unknown-elf-gdb: TUI mode is not supported
```

I build my own toolchain built from [pulp-riscv-gnu-toolchain](https://github.com/pulp-platform/pulp-riscv-gnu-toolchain), the tui mode is supported, however when I try to connect to target, there is a segmentation fault reported. Seems some patch/extra-configuration are needed for general toolchain to support cv32e40p.

Can tui mode be added in pre-built toolchain, or instruction can be given on how to build cv32e40p toolchain by ourselves.

jeremybennett, 2021-04-11:
@zarubaf I don't have the right permissions, but this needs to be assigned to @jessicamills as GNU tools lead.

The Embecosm public build bot includes the scripts for building and testing the tool chain from source.

MikeOpenHWGroup, 2021-11-04:
Picking up this stale issue.

@jessicamills is not able to access this repo - I have sent her an invite.  It is my understanding that Jessica is off until 2021-11-15.

jeremybennett, 2021-11-05:
I've asked @simonpcook, who is responsible for the pre-built tool chains, if we can enable TUI support. The original use case for the tool chains was Eclipse, for which TUI is irrelevant.