openhwfoundation/core-v-mcu #42: Core-V MCU Bring-up

zarubaf, 2021-01-25:
Migrate Content from `master`:

- Instantiate CV32E40P. A fork of the SoC is going to be necessary.
- Update documentation.

I've created two branches in this repository:

- `pulpissimo` which contains an up-to-date tree of PULPissimo.
- `core-v-mcu` which currently also contains an up-to-date tree of PULPissimo. This is the target branch to where we want to port the changes.
- Finally, there is a legacy `master` branch that contains the work of @hpollittsmith so far, as a back-up and inspiration if needed.

The idea is to re-introduce the changes to the `core-v-mcu` branch in a consistent and upstreamable manner. Finally, we are going to (try) upstream the changes to PULPissimo.

Let's use this task to break-out the different sub-tasks and track the necessary changes. So far I have:

- [ ] Fork `pulp_soc` add CV32E40P, point the commit to a stable commit within `openhwgroup/pulp_soc`.
- [ ] Adjust documentation.
- [ ] CI/CD for Genesys 2/Nexys A7

davideschiavone, 2021-01-25:
Thanks Florian.
I can take care of the pulp_soc. Let's link here issues related to those 3 points and we mark them as "done" once the relative issue has been closed so that we do not pollute this discussion with tech details

davideschiavone, 2021-01-25:
@zarubaf , can you please update your task list with a link to the following issues respectively?
#43 , #44 and #45 ?

zarubaf, 2021-01-25:
Terrific, thanks Davide. I will take care of #44.