openhwfoundation/core-v-mcu #44: CI/CD Bring-up

zarubaf, 2021-01-25:
Install continuous delivery of FPGA bitstreams for:

- Genesys 2
- Nexys A7

I've already set-up the required FPGA container. The remaining challenges are:

- [ ] Make sure PULPissimo synthesizes correctly to FPGA. I could see that support for the Nexys board is somehow broken.
- [ ] Setting up a (fork local) CI flow
- [ ] Decide whether to run the implementation flow nightly and/or on push to `master`.
- [ ] Pushing the artifacts and deciding where they should live:
    - For dedicated (stable) versions I would suggest we host them as part of the release data.
    - For nightly regressions, I suggest we host the last `n` bitstreams on something like [S3](https://aws.amazon.com/s3/?nc1=h_ls).
- [ ] Clearly link to the bitstream location in the project's README. 

Any input is very welcome.

hpollittsmith, 2021-01-29:
Hi Florian, how is the Nexys flow broken?

zarubaf, 2021-01-29:
Maybe it is a set-up issue from my side. I briefly tried the latest PULPIssimo (our `pulpissimo`) branch and it seems that:

- Somehow the IP dependencies are not working (the clock manager is not generated)
- The DRC task errors out because of missing resources I guess.

Is that something you can reproduce or is the FPGA synthesis working with the latest PULPissimo for you?

hpollittsmith, 2021-01-29:
Hi,

I built the core-v-mcu branch without any problems, but I can try the pulpissimo branch.

Hugh

zarubaf, 2021-01-29:
Okay, that is good to know. Which `nexys` board did you select?

hpollittsmith, 2021-01-29:
nexysA7-100T

zarubaf, 2021-01-29:
I see, I think I tried the 50T and that design doesn't fit. We currently don't have the board parts installed for the 100T. Let me come back to this once I've figured that out :-) Thanks for checking the issue.

zarubaf, 2021-02-01:
@hpollittsmith It seems the issue somehow persists for me. A fresh clone of this repository has problems with properly resolving the dependency to the clock manager ips:

```
# read_ip $FPGA_IPS/xilinx_clk_mngr/ip/xilinx_clk_mngr.xci
ERROR: [Vivado 12-172] File or Directory '/core-v-mcu/fpga/pulpissimo-nexys/ips/xilinx_clk_mngr/ip/xilinx_clk_mngr.xci' does not exist
```

I can manually build the clk manager with the Makefile and then the build runs fine, but that doesn't really work for any CI/CD task.

hpollittsmith, 2021-02-01:
@zarubaf what version of Vivado are you running? I've been doing clean installs and builds a fair bit over the last few weeks and haven't encountered this. I'm running 2019.2, although I want to upgrade to the latest Vivado.

MikeOpenHWGroup, 2021-11-04:
@zarubaf, @hpollittsmith what is the status of this issue?

zarubaf, 2021-11-05:
Ongoing, doesn't gate "v1.0.0"

MikeOpenHWGroup, 2021-11-12:
Hi @zarubaf and @hpollittsmith.  In a review of this issue involving @davideschiavone, @DBees and myself we determined that:
- support for Genesys 2 is desirable but not required.
- given that we have a semi-manual regression for on Nexys A7, an automated CI flow is not required.

Therefore, we are marking this issue as Closed.   Please comment if you disagree.

zarubaf, 2021-11-13:
I think we should keep it open. The issues still needs to be addressed. Although of course not fort his tapeout.