openhwfoundation/core-v-mcu #45: Write core-v-mcu Documentation

davideschiavone, 2021-01-25:
This task tracks the MCU documentation writing progress

zarubaf, 2021-01-25:
From a technical perspective:

- I don't think the latex documentation is really accessible. I would suggest to go for `mkdocs` or `readthedocs`. I think the tex sources could be easily converted to markdown using `pandoc`.
- We should check if that is something ETH is interested in upstreaming or if they want to stick to their Tex documentation.
- Potentially if we do not upstream any `core-v-mcu` related changes will be harder to track and keep out-of-tree.

Content-wise I think we should set a focus on ease of use. Not too much implementation detail but features where the SW developers would be interested:

- How to bring-up the FPGA with the pre-built bitstreams. How to connect the debugger to the board.
- Memory map and very coarse architecture manual.
- Register map of each peripheral and a description of how to get uDMA talking.

Other important information.

- Verification plan (@timsaxe has prepared something) for the tape-out
- More detailed architecture

zarubaf, 2021-01-26:
@blueww confirmed that they would be willing to accept documentation generation in static HTML. The problem apparently is the description of the memory map which comes from non-standard Excel file and needs some hacky preprocessing.

The clean solution would be to generate markdown/html from lowRISC's `reggen` tool.

blueww, 2021-01-28:
@zarubaf 
Do you need any action from me on this?

MikeOpenHWGroup, 2021-11-04:
There has not been any update to this issue for 10 months.   I've assigned it a label and an owner.

davideschiavone, 2021-11-05:
@gmartin102 - is this the latest version? https://docs.openhwgroup.org/projects/core-v-mcu/

if so we should add the link in the README and then close this issue