openhwfoundation/core-v-mcu #65: RFC: Remove `scm`

zarubaf, 2021-04-10:
Is the SCM (standard-cell memory) actually needed? Are you looking for such features @timsaxe?

@davideschiavone do you happen to know more, whats your take?

timsaxe, 2021-04-10:
Have not investigated this.  Do not have an opinion.

zarubaf, 2021-04-11:
Apparently, it wasn't used anywhere so I've removed it for now. We can add (and discuss) whether/when this is needed.

timsaxe, 2021-04-11:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Perfect.  We can let need drive us in the future.

davideschiavone, 2021-04-12:
definitively to be removed. it was part of Quentin tape out from 2017

closed 2021-04-12 by zarubaf