openhwfoundation/core-v-mcu #66: Consolidate `apb_adv_timer` and `timer_unit`

zarubaf, 2021-04-10:
I fail to understand why we would need to different timer units in the system. Can somebody enlighten me, please? @davideschiavone @timsaxe @gmartin102

gmartin102, 2021-04-10:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Florian,
Te two timer directories are unique,.  The timer_unit IP has the 2 32-bit / 1 64-bit timer that is used as the MTIMER for FreeRTOS.  The apb_adv_timer  is much more sophisticated in that it has state machines associated with it for PWM generation. I see the apb_adv_timer as the user general purpose timer resource as the timer_unit is probably;y reserved for the OS.
In the timer_unit IP, there is a redundant file timer_unit.sv which utilizes an older register interface instead of the APB bus.   I believe that file will disappear from the flattened repo structure.

Greg

zarubaf, 2021-04-11:
Great, that makes sense. I’ll keep this open until we capture that information somewhere.

MikeOpenHWGroup, 2021-11-04:
Hi @zarubaf, what is the status of this issue?

zarubaf, 2021-11-05:
No news.