openhwfoundation/core-v-mcu #67: RFC: Quicklogic modules

zarubaf, 2021-04-10:
Currently, we are missing a couple of Quicklogic modules: 

https://github.com/openhwgroup/core-v-mcu/blob/f558efb9f6e0c73ebe7ddfb678058f3ab6e4af80/rtl_list.yml#L95-L99

@timsaxe and team, I'd like to have your input on this: 

- Can we open source this? Or is it proprietory?
- If we can't how can we make a behavioral module for those components or stub the FPGA?

As soon as we have an agreement we can proceed with a further issue.

timsaxe, 2021-04-10:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

We can opensource both the fcb and math units.

https://github.com/openhwgroup/core-v-mcu/blob/f558efb9f6e0c73ebe7ddfb678058f3ab6e4af80/rtl_list.yml#L95-L99

zarubaf, 2021-04-11:
Perfect @timsaxe! That is definitely the easiest and best solution!

gmartin102, 2021-04-13:
I can add the missing files... seems like I can add to the Quicklogic fork which is called core-v-mcu-ohw in the QuickLogic-Corp area.  If this is ok or a better idea exists, let me know

closed 2021-04-16 by zarubaf