openhwfoundation/core-v-mcu #69: Check multiply-driven signal

zarubaf, 2021-04-10:
https://github.com/openhwgroup/core-v-mcu/blob/f558efb9f6e0c73ebe7ddfb678058f3ab6e4af80/rtl/core-v-mcu-soc/rtl/pulp_soc/soc_peripherals.sv#L348

Is it known that the signal is multiply driven? Would be good to check this.

zarubaf, 2021-04-10:
@timsaxe

timsaxe, 2021-04-10:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Yes, there is some commit that I am tracking down that created this and there is also another commit that breaks the interrupts.  So I’m and chasing that as well.  These are unfortunate side effects of the multi-repos system where things get checked in but the build doesn’t pull that version so I don’t notice that it created an error.  So looking forward to a single repo.

timsaxe, 2021-04-10:
Florian,

I have a working build based on master branches for QuickLogic-Corp/core-v-mcu{,-soc}, where working means that with QuickLogic-Corp/core-v-mcu-cli-tests (Eclipse/FreeRTOS) S/W project I can get UART0 and UART1 to operate using interrupts.

What I had to revert are some changes that I made to udma_subsystem which I believe caused the multiply defined signals.
The issue there is that the original PULP passed the number of peripherals (and various other things) as parameters, and I have been changing them over to defines since they are global values.  That refactoring is not completed, and apparently I broke it somehow in udma_subsystem.  I don;t believe that it is urgent to track it down.

I also had to revert some changes in pulp_soc/fc_subsystem.  The changes broke the interrupts.  I suspect the reversion makes my working version different than the version on the OHW repo, so that needs to be sorted out as part of the repo restructuring.  I'm go to investigate that today.

MikeOpenHWGroup, 2021-11-04:
Hi @zarubaf, @timsaxe.  What is the status of this issue?

MikeOpenHWGroup, 2021-11-12:
These have been resolved in the `dev` branch and will make it to `master` once #191 is merged in.

closed 2021-11-12 by MikeOpenHWGroup