openhwfoundation/core-v-mcu #108: Fix hard-coded address widths to `APB_FPGA_ADDR_WIDTH`

zarubaf, 2021-04-20:
@gmartin102 noticed a number of other places where the width is hardcoded [19:0]. They should probably be fixed to be `APB_FPGA_ADDR_WIDTH` also. Please @gmartin102 if you could address that in a separate pull request?

gmartin102, 2021-04-20:
@zarubaf I fixed a number of other address width issues in efpga_subsystem,  Also included some other logic fixes in various files and fixed a number of utility files .. openocd and  ioscript things.  I will do a pull request against Quicklogic branch address-width

zarubaf, 2021-04-22:
Please do so! Thanks!

gmartin102, 2021-04-22:
@zarubaf - I've executed a pull request, but it fails the verible format, but I can't tell why its complaining about.  I did run the format script and I had difficulting getting the git-diff script run locally so I'm not sure how to correct the errors

zarubaf, 2021-04-22:
Sorry @gmartin102 I've missed the PR. I've left some minor comments there.

MikeOpenHWGroup, 2021-11-04:
Hi @gmartin102, what is the status of this issue?

MikeOpenHWGroup, 2021-11-12:
Complete.

closed 2021-11-12 by MikeOpenHWGroup