openhwfoundation/core-v-mcu #113: Pads and pad frame

zarubaf, 2021-04-22:
When bringing up verilator I ran into two issues:

- https://github.com/openhwgroup/core-v-mcu/blob/57c8dc34f3ecbb079cf14c8562b01d73fd9f476d/rtl/core-v-mcu/top/pad_frame.sv#L130 `jtag_tdo_i` is connected to an output, I guess that is probably a problem with the IO script. @timsaxe 
- https://github.com/openhwgroup/core-v-mcu/blob/57c8dc34f3ecbb079cf14c8562b01d73fd9f476d/rtl/vendor/pulp_platform_tech_cells_generic/src/deprecated/pad_functional.sv The `pmos` and `nmos` primitives are not supported in Verilator. I would like to have your opinion on whether we can generate the pad frame (including the pads) in a new layer around `core_v_mcu` (one for the chip and one for the FPGA). That would help us entangle the situation a bit and allow for Verilator simulation.

timsaxe, 2021-04-24:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

I fixed ioscript.py to resolve the first issue. (Did this in the QuickLogic addr-width branch)

I don’t see the need to PU and PD primitives in emulation or actual silicon, and if deployed should really be a test bench feature.  SO I would suggest removing and then if anything breaks we can discuss how best to fix it.

zarubaf, 2021-04-24:
Thanks, @timsaxe for fixing 1. I'll try to tackle 2 soon and leave the issue open for that.

MikeOpenHWGroup, 2021-11-04:
Hi @zarubaf, what is the status of this issue?

zarubaf, 2021-11-05:
Out-dated

closed 2021-11-05 by zarubaf