openhwfoundation/core-v-mcu #126: Python requirements problem on Ubuntu 20.04

jeremybennett, 2021-05-18:
Running
```
pip3 install --user -r python-requirements.txt
```
on Ubuntu 20.04 gives an error
```
Successfully built edalize ipyxact ruamel.yaml pyrsistent
ERROR: edalize 0.2.5 has requirement Jinja2>=2.11.3, but you'll have jinja2 2.10.1 which is incompatible.
```
Probably need to some version matching.

MikeOpenHWGroup, 2021-11-04:
Hi @jeremybennett, I am not able to reproduce this issue.  I am running Ubuntu20.04 and the command you provide executes as expected.

Of course, this issue is somewhat stale, so maybe the version matching happened in the background.

I ran this off of core-v-mcu `c3ace54`.

jeremybennett, 2021-11-05:
@MikeOpenHWGroup It is quite likely that Ubuntu 20.04 has fixed an issue in its dependencies. I've since worked around the issue. I'm happy to close this issue if it can't be reproduced by you.

closed 2021-11-05 by jeremybennett