openhwfoundation/core-v-mcu #130: 944 warnings raised by Verilator

jeremybennett, 2021-05-19:
We build the Verilator model as follows:
```
fusesoc --cores-root . run --target=lint --setup --build openhwgroup.org:systems:core-v-mcu
```
Verilator raises 944 warnings. The warnings and frequency counts are:
```
ALWCOMBORDER     1
CASEINCOMPLETE  40
CASEX            4
CMPCONST         2
COMBDLY          2
IMPLICIT       220
LATCH            6
MULTIDRIVEN      3
PINMISSING     149
REDEFMACRO       7
STMTDLY          1
UNOPTFLAT       27
UNSIGNED         2
WIDTH          480
WIDTHCONCAT      7
```

These should all be investigated and either fixed or be marked as ignored.  Note that UNOPTFLAT will have an impact on Verilator model performance.

jeremybennett, 2021-05-19:
@MikeOpenHWGroup - sorry forgot to add this in. It is today's top of tree:
```
$ verilator --version
Verilator 4.203 devel rev v4.202-43-gaba38830
```

jeremybennett, 2021-05-19:
I did notice that you need a pretty up to date Verilator to build the MCU at all.

zarubaf, 2021-05-20:
From an RTL perspective I'd say we need to prioritize:

```
IMPLICIT       220
PINMISSING     149
MULTIDRIVEN      3
```

which, unfortunately, make up quite some amount of pins... Before we start I would also suggest we take the action to set-up CI so that we, once we fixed them, do not introduce more by adding new code!

jeremybennett, 2021-06-21:
@zarubaf This is not a valid reason for closing this issue. You've just created a waiver file to ignore all the warnings. The issue is about addressing the warnings!

MikeOpenHWGroup, 2021-11-04:
Hi @zarubaf and @jeremybennett, what is the status of this issue?

jeremybennett, 2021-11-05:
There are two aspects to this

1. Verilator warnings are useful because they indicate potential sub-optimal design and hence potential bugs. So they should be investigated, and if they are not an issue, the RTL can be tagged with Verilator comments to prevent the individual warning.

2. Some Verilator warnings are indicative of potential performance problems in the generated model. The one that stands out here is `UNOPTFLAT` but that already has its own issue [#140](https://github.com/openhwgroup/core-v-mcu/issues/140).  Some of the others may also have a small effect.

Really the warnings should be triaged into multiple issues and addressed separately, as we have done for all the `UNOPTFLAT` warnings.  I suggest keeping the issue open, but not fixing it for 1.0.0.

zarubaf, 2021-11-05:
It actually got worse. The waiver file has been compromised during merging outdated branches. I think we are back at square one with this.

MikeOpenHWGroup, 2021-11-12:
@gmartin102 has agreed to investigate this, _but_ it will not gate RTL Freeze because:
- Synopsys DC lint warnings and errors are higher priortiy (and not yet fully resolved)
- Many Verilator lint warnings are due to tool, not code issues.

matthieubraun, 2024-11-29:
Hi,

What is the current status of this issue? When I run:

```bash
fusesoc --cores-root . run --target=model-lib --setup --build openhwgroup.org:systems:core-v-mcu
```

I encounter the following error:

```bash
...
%Error: Exiting due to too many errors encountered; --error-limit=50
make: *** [Makefile:15: Vcore_v_mcu.mk] Error 1

ERROR: Failed to build openhwgroup.org:systems:core-v-mcu:0 : '['make']' exited with an error: 2
```

I assume the issue is still pending. Apart from adding more warnings to ignore in the `verilator.waiver` file or actually fixing the errors, how can one fully simulate the core-v-mcu? Has it ever been done?

commit af5328be (2021-06-14): Waive Verilator Warnings (#138)

* Fix REDEFMACRO lint warning

* Waive Verilator warnings (fixes #130)