openhwfoundation/core-v-mcu #150: Debug module reports the wrong number of Harts

jeremybennett, 2021-07-15:
The MCU has a single hart. The debug unit does not report this.

Section 3.3 of the ratified RISC-V Debug standard (v 0.13) says that to enumerate the harts:
> Select each hart starting from 0 until either `anynonexistent` in `dmstatus` is 1 or the highest index (based on `HARTSELLEN`) is reached.

If we do this for the MCU we get:
```
hart =       0, dmcontrol = 0x00000001, dmstatus = 0x000c0c82
hart =       1, dmcontrol = 0x00010001, dmstatus = 0x000c0c82
...
hart =       6, dmcontrol = 0x00060001, dmstatus = 0x000c0c82
hart =       7, dmcontrol = 0x00070001, dmstatus = 0x000c0c82
hart =       8, dmcontrol = 0x00080001, dmstatus = 0x00003082
hart =       9, dmcontrol = 0x00090001, dmstatus = 0x00003082
...
hart =    1022, dmcontrol = 0x03fe0001, dmstatus = 0x00003082
hart =    1023, dmcontrol = 0x03ff0001, dmstatus = 0x00003082
hart =    1024, dmcontrol = 0x00000041, dmstatus = 0x000ccc82
```
In other words
- harts 0-7 are shown as being existent and available
- harts 8-1023 are shown as being existent, but unavailable
- harts 1024 upwards are shown as being non-existent

The troublesome RTL is in [line 29 of soc_domain.sv](https://github.com/openhwgroup/core-v-mcu/blob/master/rtl/core-v-mcu/top/soc_domain.sv#L29) which sets parameter `NB_CORES` to 8, and [line 344 of soc_domain.sv](https://github.com/openhwgroup/core-v-mcu/blob/master/rtl/core-v-mcu/top/soc_domain.sv#L344) which sets localparam `NrHarts` to 1024.

I think this is a misunderstanding of the standard. I understand the PULP debug module is designed to support up to 1024 cores, should be setting `HARTSELLEN` to 10, and then marking any cores that are not implemented as non-existent (_not_ unavailable).

In the case of CORE-V MCU, this means just a single core. For now the Verilator model for debugging will override the reported number of cores.

gmartin102, 2021-07-15:
So my question is should both the NB_CORES and NrHarts be set to 1?  since that is all that we are implementing thta would seem to be the correct interpretation.

jeremybennett, 2021-07-16:
That seems plausible, but it would be good to understand if they have other uses within the debug unit. I haven't examined the RTL in detail, @davideschiavone pointed me to the code.

BTW, I think it is also an error in PULP, which should be setting HARTSELLEN to 10 (indicating they could configure for up to 10^2 harts) and then mark all except the first 8 (since PULP has 8 cores) as non-existent.  Arguably we should also set HARTSELLEN for CORE-V MCU, since presumably it could only accomodate up to 1024 cores in the future. But that is less important.

jeremybennett, 2021-08-18:
I notice that with the latest commit [145a458](https://github.com/openhwgroup/core-v-mcu/commit/145a458344ce14b4cc0858ded3e684b73a296e97) in the master branch. `HARTSELLEN` is now reported as 0 and there is a single Hart shown as available.

Marking as closed. Thanks.

closed 2021-08-18 by jeremybennett