openhwfoundation/core-v-mcu #155: Reset on debug connect

zarubaf, 2021-08-09:
OpenOCD does not reliably connect. We found a sequence that usually works:
- turn on the board;
- start up OpenOCD;
- after OpenOCD has started, but before it times out hit CPU reset button.

> I suspect that this might be related to the reset of the debug module, which will reset the CPU core once connected using `ndmreset`. This needs to be bisected further, but the debug module should definitely be capable of resetting its `harts`.

gmartin102, 2021-08-09:
@florian -- I have not seen this issue.  I use openocd many times per day and the only issue i have ran into is an occasional lost Openocd process that ties up the USB port from a poor termination...(usually from downloading a bitstream while openocd and gdb are connected to the target). I'll look into ndmreset a bit as I have noticed that the reset button while openocd is connected does  send the system off the deep end

zarubaf, 2021-08-10:
Okay, that's good to know. Was just my first guess. Somehow there seems to be trouble with others reproducing a reliable connection.

MikeOpenHWGroup, 2021-11-04:
Hi @zarubaf, what's the status of this issue?

mbaykenar, 2023-02-16:
Hi,

I am trying to debug core-v-mcu on Nexys A7

The problem is somehow related to this: https://github.com/openhwgroup/core-v-mcu/issues/155#issuecomment-895465455

Sometimes OpenOCD can't access debug port, sometimes no problem, I am also trying to find a sequence that always works.

I am also not sure that I am doing right thing.

I connect boards's PROG/UART to PC and download the bitstream
Then I open a terminal and see A2 BOOTME
Then I connect digilent HS2 and try to debug
When debugging starts (if I am luck), I can't access terminal that sending A2 BOOTME so that I can run cli_test commands

is there any impropriety in that sequence?

Regards

MikeOpenHWGroup, 2023-02-18:
Hi @mbaykenar, thanks for sticking with this.  Have you looked at the [Quick Start Guide](https://github.com/openhwgroup/core-v-mcu/blob/master/emulation/quickstart/README.md)?  This will lead you through the process step-by-step.

A couple of things I could suggest:
- In general, I have my HS2 connected to the debug port right from the start.
- The JTAG port is not enabled unless SW0 is in the "up" position.  As shown [here](https://github.com/openhwgroup/core-v-mcu/blob/master/emulation/quickstart/README.md#nexys-a7---digilent-hs2-connection), SW0 is on the lower-right-hand-corner of the Board.

mbaykenar, 2023-02-20:
Hi @MikeOpenHWGroup 

I don't remember how many times I looked and went over the QSG, but still forgot changing the position of SW0 even though there is a bold typed **NOTE** there :l 

I re-tried with correct switch and now it worked as expected! Thank you very much.

But maybe I can say a different behavior: Even though the switch is in low position, sometimes I can debug the core, analyze registers etc. Maybe, this is due to being master of JTAG port? In Xilinx SoC's for example, even though you don't select JTAG as boot option, you can still debug ARM core in Zynq. I don't know but wanted to report the situation.

Regards,

MikeOpenHWGroup, 2023-02-20:
Yes, we have _all_ been bitten by that one.  Multiple times in my case.

> Even though the switch is in low position, sometimes I can debug the core

Is your debug port on JB (PMOD on lower right), or J6 (micro-USB on upper left)?