openhwfoundation/core-v-mcu #159: Accessing FP CSRs crashes the debug unit

jeremybennett, 2021-08-18:
I am running a Verilator model with the standard configuration, i.e. no FPU and no PULP extensions.

If I read one of the FPU CSRs, then I will get the value last read from a CSR and all subsequent reads to CSRs will return the same value.  Here is a demonstrator:
```
$ ./testbench.exe 
Timescale 1ns / 1ns
IDCODE = 0x249511c3
  Version:  0x2
  Part #:   0x4951
  Manuf ID: 0x0e1
    JEP106 ID:      0x61
    Continuation #: 0x1
DTMCS = 0x00001071
  dmihardreset: 0
  dmireset:     0
  idle:         1
  dmistat:      0
  abits:        7
  version:      1
  Addr mask:    0x000000000000007f
Max HARTS: 0x00000001
Testing HART availability

Hart 0, dmstatus: 0x000c0c82 [ impebreak = false, havereset = true, resumeack = false, nonexistent = false, unavail = false, running = true, halted = false, authenticated = true, authbusy = false, hasresethaltreq = false, confstrptrvalid = false, version = 2 ]

Num HARTS: 1

Hart 0
  hartinfo: 0x00212380 [ nscratch = 2, dataaccess = true, datasize = 2, dataaddr = 0x380 ]
  abstractcs: 0x08000002 [ progbufsize = 8, busy = false, cmderr = 0 (none), datacount = 0x2 ]

Hart 0 halted
PC = 0x1a0001bc

cycle = 0x000013dc
instret = 0x0000051b
PC = 0x1a0001bc

fflags = 0x1a0001bc
cycle = 0x1a0001bc
instret = 0x1a0001bc
PC = 0x1a0001bc
```

Note that reading the hardware loop CSRs does not have this effect, even though hardware loop is not connected.

I suggest that reading a non-existent CSR should be an error, possibly even raising an exception. But it should not torch the debug unit.

jeremybennett, 2021-08-18:
An additional observation that attempting to access the FP registers has the same impact as accessing the FP CSRs.

MikeOpenHWGroup, 2021-08-18:
@jeremybennett, can you raise an issue on core-v-verif about this?  I am 99% certain we tested this on the CV32E40P, but I'd like a reminder to double check that (I'm on the road at present and will dutifully forget all about it without an issue to force me to look into it.)

Thanks

jeremybennett, 2021-08-19:
Also now filed as https://github.com/openhwgroup/core-v-verif/issues/820, which has detailed instructions for reproducing.

MikeOpenHWGroup, 2021-11-04:
Assigning to myself as I am also looking at [core-v-verif #820](https://github.com/openhwgroup/core-v-verif/issues/820)