openhwfoundation/core-v-mcu #168: IDCODE change to 0x10001602 causes OPENOCD to fail

gmartin102, 2021-09-05:
Openocd reports an invalid IDCODE of 0x10001602.  Seems It wants at least a "1" in the LSB position and the manufacturer code is illegal.

zarubaf, 2021-09-05:
Indeed, I remember the LSB should be 1. I’ll fix that. For the other thing we need to patch openocd with the OpenHW IDCODE.

gmartin102, 2021-09-07:
@zarubaf -- for a temporary fix, I am going to make the IDCODE 0x10001603 to make openocd happy, but That Manufacture code (0x301) seems to be assigned to MOVEKING -
Info : JTAG tap: riscv.cpu tap/device found: 0x10001603 (mfg: 0x301 (MOVEKING), part: 0x0001, ver: 0x1)
Do we have or need a Manufacture code or is that for the end users to assign?

zarubaf, 2021-09-07:
Sorry @gmartin102: So our JEDEC code is `0x602` (as registered in JEP106). I think my misconception was that I thought the debug module would shift that appropriately. So the proper ID CODE should be:

`0x10001C05`

Can you check that, there should probably be unknown vendor and I will get the PR to OpenOCD (once I've figured out how that works).

[image]

zarubaf, 2021-09-07:
Can you try quickly and report back please?

gmartin102, 2021-09-07:
Yes, will get back to you as soon as i can complete a P&R

gmartin102, 2021-09-07:
@zarubaf 0x10001c05 works --
Info : JTAG tap: riscv.cpu tap/device found: 0x10001c05 (mfg: 0x602 (<unknown>), part: 0x0001, ver: 0x1)
Info : datacount=2 progbufsize=8
Info : Examined RISC-V core; found 1 harts

gmartin102, 2021-09-07:
Yes I will make a pr from dev to master with that corrected and an additional correction to the ioscript later today

MikeOpenHWGroup, 2021-11-04:
@gmartin102, can this issue be closed?

zarubaf, 2021-11-05:
Added to OpenOCD

closed 2021-11-05 by zarubaf