openhwfoundation/core-v-mcu #170: Silicon validation results

navaneeth-cirel, 2021-09-08:
Hi,

Are there any test results from the silicon which was taped out for core-v-mcu?

Thanks

MikeOpenHWGroup, 2021-09-08:
Thanks for your interest @navaneeth-cirel.  The core-v-mcu has not yet taped-out (rest assured we are working on it and will make a big deal about it when it does).

What types of test results are you interested in?

navaneeth-cirel, 2021-09-08:
Hi @MikeOpenHWGroup,

Thank you for a quick reply. We are mainly interested in the power numbers (active, sleep etc.) similar to what is seen in a typical Microcontroller datasheet and also any benchmarking scores (e.g. Coremark). The main reason is that we are planning to use the CV32E40P core in our SoC and would like get some ballpark figures of the power and performance that can be expected.
It would be great if you can point me to the results of netlist simulation of the above as well (if available).

Thanks,
Navaneeth

MikeOpenHWGroup, 2021-09-08:
If you are looking for CV32E40P-specific information, you will not find that here as this repo is for the MCU.

OpenHW has published some performance data for the CV32E40P which you can find [here](https://github.com/openhwgroup/core-v-docs/blob/master/cores/performance/ppa_data.xlsx) in our core-v-docs repository.  Admittedly that is a terrible place for it and I'll open an issue to see if we can't make that available on the [CV32E40P User Manaual](https://docs.openhwgroup.org/projects/cv32e40p-user-manual/intro.html).

MikeOpenHWGroup, 2021-11-04:
Hi @navaneeth-cirel, has your question been answered?  If I do not hear from you before 2021-11-11, I will close this issue.

navaneeth-cirel, 2021-11-05:
Hi @MikeOpenHWGroup ,

I feel this is partially answered, I got some performance numbers from the [link](https://github.com/openhwgroup/core-v-docs/blob/master/cores/performance/ppa_data.xlsx) you shared but I was unable to find any power numbers. Moreover I was looking more for numbers from the core-v-mcu as a whole to get idea from a typical system level perspective.

MikeOpenHWGroup, 2021-11-05:
Understood @navaneeth-cirel.  These will be published once we have the silicon in hand.  You can find the current schedule in the MCU Project Approval document ([link](https://github.com/openhwgroup/core-v-docs/blob/master/program/Project%20Descriptions%20and%20Plans/CORE-V%20MCU%20SoC/PA%20document%20Oct%2025%202021.md#summary-of-timeline)).