openhwfoundation/core-v-mcu #181: Some problems when make buildsim sim

kangliyu1, 2021-10-14:
[image]
I run the command make buildsim sim, and the problem as shown in the figure appeared in questasim

kangliyu1, 2021-10-15:
> It is not attached to this issue. You must drag-and-drop from the web-browser, not attach it to email.
[logifle.txt](https://github.com/openhwgroup/core-v-mcu/files/7350217/logifle.txt)

gmartin102, 2021-10-15:
The file you included is the logfile from the modelsim command.  I am interested in seeing if the uartdpi library was compiled properly. after you run ' make buildsim' you should have a file called uartdpio.so in the core-v-mcu/tb/uartdpi directory.
That library is linked with the rtl to provide a /dev/pts uart to send/receive data to the simulation.  the shared library is compiled in the buildsim portion, but it appears that VSIM is attempting to also compile it.  it maybe due to a newer version of QuestaSim.  You could remove the two uarts from the testbench core-v-mcu/tb/core_v_mcu_tb.sv and the associated -sv_lib option on the vsim command line to see the simulation progress.

kangliyu1, 2021-10-18:
@gmartin102 Hello, do you mean to change like this? , I made another mistake after changing it like this
[image]
[image]
[log.txt](https://github.com/openhwgroup/core-v-mcu/files/7361638/log.txt)