openhwfoundation/core-v-mcu #187: FP 0 reported through debug unit as tied to constant positive 0

jeremybennett, 2021-11-03:
The verilator model under development for the CORE-V MCU has been used to access the FPU registers. It appears that register `f0`, which should be one of eight temporaries is tied to constant 0x0 (IEEE 754 positive 0).  Details of the reproducer and VCD are given below. The output from the Verilator model test framework is as follows:
```
$ ./testbench.exe --test-fprs --vcd fpr-access.vcd
Timescale 1ns / 1ns
TOP.core_v_mcu.i_soc_domain.l2_ram_i.bank_sram_pri0_i.u0
TOP.core_v_mcu.i_soc_domain.l2_ram_i.bank_sram_pri1_i.u0
TOP.core_v_mcu.i_soc_domain.l2_ram_i.CUTS[0].bank_i.u0
TOP.core_v_mcu.i_soc_domain.l2_ram_i.CUTS[1].bank_i.u0
TOP.core_v_mcu.i_soc_domain.l2_ram_i.CUTS[2].bank_i.u0
TOP.core_v_mcu.i_soc_domain.l2_ram_i.CUTS[3].bank_i.u0
IDCODE = 0x10001c05 [version = 0x1, part # =  0x0001, manuf ID: 0x602 [JEP106 ID = 0x02, continuation # = 12]]
DTMCS = 0x00001071 [dmihardreset = false, dmireset = false, idle = 1, dmistat = 0, abits = 7, version = 1, addr mask = 0000007f]
Max HARTS: 0x00000001
Testing HART availability

Hart 0, dmstatus: 0x000c0c82 [ impebreak = false, havereset = true, resumeack = false, nonexistent = false, unavail = false, running = true, halted = false, authenticated = true, authbusy = false, hasresethaltreq = false, confstrptrvalid = false, version = 2 ]

Num harts: 1

Testing hart 0
==============

Reading FPR values:
  ft0 (f0)   = 00000000  ft1 (f1)   = 1a0000ae  ft2 (f2)   = 1c0003a0
  ft3 (f3)   = 00000000  ft4 (f4)   = 00000000  ft5 (f5)   = 00000000
  ft6 (f6)   = 00000000  ft6 (f7)   = 00000000  fs0 (f8)   = 1a000244
  fs1 (f9)   = 00000000  fa0 (f10)  = 00000208  fa1 (f11)  = 00000001
  fa2 (f12)  = 1c0003ac  fa3 (f13)  = 00000000  fa4 (f14)  = 00000001
  fa5 (f15)  = 1a102100  fa6 (f16)  = 00000000  fa7 (f17)  = 00000000
  fs2 (f18)  = 00000000  fs3 (f19)  = 00000000  fs4 (f20)  = 00000000
  fs5 (f21)  = 00000000  fs6 (f22)  = 00000000  fs7 (f23)  = 00000000
  fs8 (f24)  = 00000000  fs9 (f25)  = 00000000  fs10 (f26) = 00000000
  fs11 (f27) = 00000000  ft8 (f28)  = 00000000  ft9 (f29)  = 00000000
  ft10 (f30) = 00000000  ft11 (f31) = 00000000
Writing FPR values:
ft0 (f0):   Wrote: 0xffffffff, read back: 0x00000000
ft0 (f0):   Wrote: 0xdeadbeef, read back: 0x00000000
```
## Reproducer

The Verilator model can be built as follows. **Note:** this requires Verilator 4.200 or later and a reasonably modern GCC running under Linux (I've not tested it on Cygwin or MSys under Windows).

1. Clone this [CORE-V MCU repository](https://github.com/openhwgroup/core-v-mcu);
2. Build the model library by running `make model-lib` from the top level
3. Clone my fork of the [Embench target repository](https://github.com/jeremybennett/embdebug-target-core-v).
4. Check out the `jpb-jtag-driver` branch
5. Change to the `target` directory
6. Edit the `Makefile` and change the value of `MCU_DIR` on line 23 to point to the location of the CORE-V MCU repository checked out in step 1 above.
7. Run `make`
8. This should build an executable, `testbench.exe`
9. Run the command `./testbench.exe --test-fprs --vcd fpr-access.vcd` which will dump the values in each FPR and then attempt to write and read values to/from each register (see output above). The VCD dump will be created at the end of execution.

Other features can be tested - run `testbench.exe --help` to find out more.

## VCD file

The VCD file for the above example is 21GB, or 232MB compressed with xz, so too large to add to GitHub easily. I'll work out where to put it tomorrow and post a link here.

jeremybennett, 2021-11-04:
Compressed VCD file is available for download: [fpr-access.vcd.xz](https://files.embecosm.com/s/xfAwQ9K5S3aXpn7)

MikeOpenHWGroup, 2021-11-12:
@jeremybennett, once #191 is merged in, the MCU will not support a floating point unit.  Therefore, this issue is "won't fix" for V1.0.0, so I have removed the "Gate RTL Freeze" label.   This issue can be revisited if/when a future MCU incorporates the FP unit.

jeremybennett, 2021-11-23:
Seems reasonable to  me.