openhwfoundation/core-v-mcu #188: Centralize compiler macros in a top-level package

MikeOpenHWGroup, 2021-11-04:
There is at least one instance of a compiler macro in CORE-V-MCU that is defined and used in disjointed parts of the RTL.  The macro in question is `USE_FPU`.  It is used in  [fc_subsystem.sv](https://github.com/openhwgroup/core-v-mcu/blob/master/rtl/core-v-mcu/fc/fc_subsystem.sv#L156) to set the FPU parameter for the CV32E40P core.  The macro is defined in [pulp_soc_defines.sv](https://github.com/openhwgroup/core-v-mcu/blob/master/rtl/includes/pulp_soc_defines.sv#L79).  This is a dangerous situation because the value of `USE_FPU` passed to the core is dependent on compile order:
- if `pulp_soc_defines.sv` is compiled before `fc_subsystem.sv` then USE_FPU is 1.
- if `fc_subsystem.sv` is compiled before `pulp_soc_defines.sv` then USE_FPU is undefined.

At a minimum, the above situation must be resolved.   It would also be a great idea to review all macros in the MCU and ensure that either the macros are defined in a top-level package or define in the module in which they are used.

@gmartin102, I am assigning this to you as you are the default RTL engineer on this project.  Feel free to delegate.

gmartin102, 2021-11-04:
@MikeOpenHWGroup - I thought the `include "pulp_soc_defines.sv" on line 10 or so inthe fc_subsystem would read the definitions in before compilation.   maybe the file should be an svh suffix to properly denote it as a header file?  Our (My) intent was to avoid the compile order issue and eliminate the confusion by having the parameter ripple up a down the hierarchy as an instantiated  parameter.

MikeOpenHWGroup, 2021-11-04:
> the `include "pulp_soc_defines.sv" on line 10 or so inthe fc_subsystem would read the definitions in before compilation.

Yeah it "works", but its dangerous to make the `fc_subsystem` dependent on the `pulp_soc_defines`.

> maybe the file should be an svh suffix to properly denote it as a header file? 

I like that one better. If we are going to have defines anywhere they should be in a header file (svh) at the top of the hierarchy.

Better yet, just get rid of it.  the `fc_subsystem` already has a `USE_FPU` parameter so that should be passed to the core.

MikeOpenHWGroup, 2021-11-12:
Much, but not all, of this work has already been completed.  Completion of this task will not gate V1.0.0