openhwfoundation/core-v-mcu #189: Incorrect parameterization of CV32E40P in CORE-V-MCU

MikeOpenHWGroup, 2021-11-04:
Within the CORE-V-MCU, the CV32E40P core is instantiated in [fc_subsystem.sv](https://github.com/openhwgroup/core-v-mcu/blob/master/rtl/core-v-mcu/fc/fc_subsystem.sv#L155) as follows:
```
  cv32e40p_core #(
      .FPU(`USE_FPU),
      .PULP_XPULP(1)
  ) lFC_CORE (
...
```
There are several issues here:
1. The `USE_FPU` macro (tick-define) is defined in `pulp_soc_defines.sv` (see #188).
2. Assuming that the intended value of `USE_FPU` is 1, this is not a Release 1.0.0 (v1) configuration.
3. `PULP_XPULP` is set to 1, which is not a Release 1.0.0 configuration.

The [CV32E40P User Manual](https://cv32e40p.readthedocs.io/en/latest/integration/#core-integration) makes it clear how to instantiate the v1 version of this core and the CORE-V-MCU is not doing that.   This is a problem because the [MCU Project Launch](https://cv32e40p.readthedocs.io/en/latest/integration/#core-integration) document indicates that the MCU is to use the Release 1.0.0 version of the core.  This decision is reinforced in the [MCU Plan Approved](https://cv32e40p.readthedocs.io/en/latest/integration/#core-integration) document:
```
The MCU project will bring the CV32E40P IP to TRL6 from a starting point at TRL5.
```
Only Release 1.0.0 of the CV32E40P is at TRL5.

This issue can be resolved when the instantiation parameters of the CV32E40P in the MCU are set as per the Release 1.0.0 values in the User Manual.

MikeOpenHWGroup, 2021-11-12:
Fixed in #191

gmartin102, 2021-11-18:
closed - parameterization is corrected per cv32e40p documentation

closed 2021-11-18 by gmartin102