openhwfoundation/core-v-mcu #194: `dm_exception_addr_i` hardwired to 0x0

MikeOpenHWGroup, 2021-11-11:
As of `c3ace54`, the top-level primary input to the CV32E40P core in the core-v-mcu `dm_exception_addr_i[31:0]` is hardwired to 0x0 in the MCU ([link](https://github.com/openhwgroup/core-v-mcu/blob/master/rtl/core-v-mcu/fc/fc_subsystem.sv#L167)).

As discussed in [core-v-verif issues #820](https://github.com/openhwgroup/core-v-verif/issues/820#issuecomment-966093972), `dm_exception_addr_i` should be set to the "appropriate address in the Debug Module".

It is not clear to me what the precise definition of "appropriate address in the Debug Module" is.  From the core's perspective it is just a 32-bit number since the CV32E40P has a flat 32-bit wide memory space.  Within the MCU the "appropriate address in the Debug Module" could be a fixed address in physical memory, or it could be an address in within the debug module itself.

davideschiavone, 2021-11-11:
@bluewww, @gmartin102 do we have an exception routine inside the Debug ROM or somewhere in the Software?
This is the address the core jumps when in DEBUG MODE, and an **_illegal instruction_**, or an **_ecall_**, or an **_mret_**  is encountered.
Pining also @jeremybennett to know how this is handled in SW

gmartin102, 2021-11-11:
@davideschiavone, yes i did a little investigation and the debug rom does have an exception routine (_exception:) that is for exception processing while in debug mode.  I believe the address is 0x1a11080c, so I plan to try that address and make sure nothing breaks but i'm not to sure how to actually test it.   maybe @jeremybennett has some idea on how to test it

davideschiavone, 2021-11-11:
ok cool, then if that is the case we can update the RTL.

davideschiavone, 2021-11-11:
the software team should decide whether how the exception should be treated. It can also become programmable by adding a register in the APB-domain that is written by the crt0, ROM, or jtag loader. Of course, it's not the safest option but it is one option.

gmartin102, 2021-11-18:
Fixed - dm_execption address set to 0x1a11080c as specified in dm module rom code.

closed 2021-11-18 by gmartin102