openhwfoundation/core-v-mcu #212: `make model-lib` is broken

jeremybennett, 2022-02-01:
Commit [31defb5edd13064362e23d91bdbe2aacd89fbe8a](https://github.com/openhwgroup/core-v-mcu/commit/31defb5edd13064362e23d91bdbe2aacd89fbe8a) broke `make model-lib` (used by the Verilator modeling project).
```
$ make model-lib
fusesoc --cores-root . run --target=model-lib --setup \
	--build openhwgroup.org:systems:core-v-mcu | tee model-lib.log
INFO: Preparing openhwgroup.org:ip:apb2apbcomp:0
INFO: Preparing openhwgroup.org:ip:apb2per:0
INFO: Preparing openhwgroup.org:ip:apb_adv_timer:0
INFO: Preparing openhwgroup.org:ip:apb_fll_if:0
INFO: Preparing openhwgroup.org:ip:apb_gpio:0
INFO: Preparing openhwgroup.org:ip:apb_i2cs:0
INFO: Preparing openhwgroup.org:ip:apb_node:0
INFO: Preparing openhwgroup.org:ip:apb_timer_unit:0
INFO: Preparing openhwgroup.org:ip:axi_slice:0
INFO: Preparing openhwgroup.org:ip:cv32e40p:0
INFO: Preparing openhwgroup.org:ip:logint_dc_fifo_xbar:0
INFO: Preparing openhwgroup.org:ip:tcdm_interconnect:0
INFO: Preparing openhwgroup.org:ip:udma_camera:0
INFO: Preparing openhwgroup.org:ip:udma_core:0
INFO: Preparing openhwgroup.org:ip:udma_external_per:0
INFO: Preparing openhwgroup.org:ip:udma_filter:0
INFO: Preparing openhwgroup.org:ip:udma_i2c:0
INFO: Preparing openhwgroup.org:ip:udma_i2s:0
INFO: Preparing openhwgroup.org:ip:udma_qspi:0
INFO: Preparing openhwgroup.org:ip:udma_sdio:0
INFO: Preparing openhwgroup.org:ip:udma_uart:0
INFO: Preparing pulp-platform.org::common_cells:1.20.0
INFO: Preparing pulp-platform.org::tech_cells_generic:0
INFO: Preparing quicklogic.com:ip:efpga:0
INFO: Preparing openhwgroup.org:ip:l2_tcdm_hybrid_interco:0
INFO: Preparing pulp-platform.org::axi:0.28.0
INFO: Preparing pulp-platform.org::fpnew:0
INFO: Preparing pulp-platform.org::fpu_div_sqrt_mvp:0
INFO: Preparing pulp-platform.org::riscv_dbg:0
INFO: Preparing openhwgroup.org:systems:core-v-mcu:0
ERROR: Setup failed : Cannot find tb/mem_init_files/verilatorBoot.mem in :
	.
	.
```

gmartin102, 2022-02-01:
Fixed the Verilator issue.. inclusion of PLL logic broke the clock generation for Verilator and somehow the verilator boot code was deleted from the repo.

jeremybennett, 2022-02-01:
@gmartin102  Tested and this fixes the model building issue. Thanks for the quick patch.

Now looks like the debug unit is broken again. Symptoms seem very similar to previously. I'll raise this as a separate issue.

jeremybennett, 2022-02-01:
With the latest push, `make model-lib` works. However when I then include the verilator model in its test wrapper, the debug unit is broken and reports no harts. Documented in issue [#213](https://github.com/openhwgroup/core-v-mcu/issues/213).

gmartin102, 2022-02-01:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Jeremy,
My mistake on fat fingering on of the updated files to include on the previous push.  I know this makes you're testing a pain to have to keep trying this build.  This highlights the weakness of Verilator on this type of whole chips simulation.  The actual device has a real PLL that has relatively complicated gating an control.  The Verilator build causes quite a disruption to the clock generation and doesn’t track the actual behavior of the device and therefore I don’t spend a proper amount of time verifying its functionality.  I’m sorry that a previous push to the master branch wasn’t tested better and I see I need to keep you in the review loop to make sure I don’t miss these things.

Greg

gmartin102, 2022-02-01:
VerilatorBoot.mem file was missing from the repo. This file is an alternate bootloader for use with verilator .

MikeOpenHWGroup, 2022-02-02:
Hi @gmartin102, let's follow the work-flow whereby the originator of an issue (@jeremybennett in this case), determines whether or not to close an issue.

jeremybennett, 2022-02-02:
Happy to close this issue. The broken debug unit it covered in #213

MikeOpenHWGroup, 2022-02-17:
Hi @gmartin102 and @jeremybennett.  As near as I can tell, this issue appears to _not_ have been resolved.  When attempting to compile with either Verilator or Vivado, I get:
`ERROR: Setup failed : Cannot find tb/mem_init_files/verilatorBoot.mem in :`.

What did you do to resolve this issue?

gmartin102, 2022-02-17:
This is fixed in the genesys2 branch.. I'm not sure who should do this merge back to master.  I seem to mess things up to often so I mentioned that we needd to spend a little time to get things organized.  Jeremy tested in the genesys2 branch ithink

jeremybennett, 2022-02-17:
That is correct - I tested the `genesys2` branch.

I had assumed once the branch had passed review, it would be merged into master. @gmartin102 do you need some help with this?

MikeOpenHWGroup, 2022-02-17:
Ah, got it.

@gmartin102, I'll reach out to you today and we can merge up the `genesys2` branch onto the `master`.  I see pull-request #211 is already in place, so we can review it together.  Shouldn't take more than a half hour.

jeremybennett, 2022-03-11:
@mike as soon as the `genesys2` branch is merged this issue can be closed.

closed 2022-03-12 by MikeOpenHWGroup