openhwfoundation/core-v-mcu #213: Debug unit reports no harts

jeremybennett, 2022-02-01:
I have tested the latest work with the `genesys2` branch (commit 138c298).

The Verilator model builds using `make model-lib`. However when the model is included in the test harness, the debug unit is now reporting no harts. I built the Verilator test harness (`embench-target` repository, `target` directory, `jpb-jtag-driver` branch)
```
$ cd target
$ make clean
$ make
```

I then run the testbench:
```
$ ./testbench.exe --test-status --test-gprs --test-fprs --test-csrs
Timescale 1ns / 1ns
TOP.core_v_mcu.i_soc_domain.l2_ram_i.bank_sram_pri0_i.u0
TOP.core_v_mcu.i_soc_domain.l2_ram_i.bank_sram_pri1_i.u0
TOP.core_v_mcu.i_soc_domain.l2_ram_i.CUTS[0].bank_i.u0
TOP.core_v_mcu.i_soc_domain.l2_ram_i.CUTS[1].bank_i.u0
TOP.core_v_mcu.i_soc_domain.l2_ram_i.CUTS[2].bank_i.u0
TOP.core_v_mcu.i_soc_domain.l2_ram_i.CUTS[3].bank_i.u0
IDCODE = 0x10001c05 [version = 0x1, part # =  0x0001, manuf ID: 0x602 [JEP106 ID = 0x02, continuation # = 12]]
DTMCS = 0x00001071 [dmihardreset = false, dmireset = false, idle = 1, dmistat = 0, abits = 7, version = 1, addr mask = 0000007f]
Max HARTS: 0x00100000
Testing HART availability

Warning dmiWrite retry requested
Warning dmiRead retry requested
Hart 0, dmstatus: 0x03ffffc1 [ impebreak = true, havereset = true, resumeack = true, nonexistent = true, unavail = true, running = true, halted = true, authenticated = true, authbusy = true, hasresethaltreq = false, confstrptrvalid = false, version = 1 ]

Num harts: 0
```
The messages about retry requested are worrying, and ISTR we saw the same when we had problems previously.

timsaxe, 2022-02-01:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.
>
> Issue is fixed by correcting the PLL simulation model. Verilator clock generation is significantly different from soc and emulation clocks and the inclusion of the PLL model failed to source the verilator clocks correctly.

Do we need to add something to either CI or regression to catch this earlier?

MikeOpenHWGroup, 2022-02-02:
Hi gang - a few comments on this one:
1. Please push changes from your local fork to the `dev` branch of this repo.
2. We should not merge anything up to the master branch until it has successfully passed regressions in simulation, emulation and Verilator.
3. Please let the originator of an issue be the one to close it.   @jeremybennett is the originator of this issue, so I will reopen it for his comment.

jeremybennett, 2022-02-02:
@gmartin102 Thanks for the quick turnaround. Tested and all working sweetly.

closed 2022-02-02 by jeremybennett