openhwfoundation/core-v-mcu #217: What is src_files.yml used for?

MikeOpenHWGroup, 2022-03-09:
Hi @davideschiavone, what is the [src_files.yml](https://github.com/openhwgroup/core-v-mcu/blob/master/rtl/core-v-mcu/src_files.yml) used for?

davideschiavone, 2022-03-11:
it was used as part of PULPissimo as a manifest file. The iptools would generate compile and analyse scripts. core-v-mcu uses fusesoc so src file is deprecated

MikeOpenHWGroup, 2022-03-11:
Can we safely remove them?

davideschiavone, 2022-03-11:
yes but we should not remove by hands files in the vendor folder - only outside the vendorized IP - instead, with you should exclude them from the source (e.g. https://github.com/openhwgroup/core-v-mcu/blob/master/rtl/vendor/openhwgroup_cv32e40p.vendor.hjson#L15) 

so we should:

1. Do a big clean up of the things we do not use, e.g. iptools, src files, rtl and ip lists, vcompile folder, etc. Here we need help also from @gmartin102 to be sure we are not removing things that we are instead using
2. Update the vendor hjson files to remove things we do not use, as for e.g. the src_files.