openhwfoundation/core-v-mcu #226: [Doc] toolchain indication in the quickstart guide

suppamax, 2022-04-27:
In the Quickstart Guide I would expect to find some indication about the toolchain to compile the SW.

MikeOpenHWGroup, 2022-04-27:
There are two reasons why you will not find that in the QSG:
1. The QSG assumes the user is going to load a pre-built srec file, so no access of the toolchain is required.  For those who plan to use the Eclipse IDE, there is a link at the bottom of the QSG that takes you to a place that explains how to setup the toolchain for Eclipse to use.
2. The software (cli_test) is hosted in a separate repo [core-v-mcu-cli-test](https://github.com/openhwgroup/core-v-mcu-cli-test). Admittedly this is not clear in the READMEs or the QSG.

124107157-KV, 2026-08-19:
Hi, I'd like to work on this issue. I checked the current `master branch` and the Quick Start Guide still does not explicitly explain the software toolchain requirements.

I plan to update the Quick Start Guide to clarify that the supplied pre-built `cli_test.srec` can be used without installing a compiler toolchain, while users who want to build or modify `cli_test` from source need the CORE-V GCC toolchain. I will also point users to the `core-v-mcu-cli-test` repository where the software and toolchain setup instructions are maintained.

I'll submit a documentation PR for this.

closed 2026-08-21 by MikeOpenHWGroup