openhwfoundation/core-v-mcu #233: name 'N_IO' is not defined

kyeoh2, 2022-06-11:
Any idea why I have this error?  I try to download and run make src, and I run into this error, any advice will be appreciated.

kyeoh1@ubuntu:/work/kyeoh1/core-v-mcu$ make src
make: Circular rtl/includes/pulp_peripheral_defines.svh <- rtl/includes/pulp_peripheral_defines.svh dependency dropped.
make: Circular rtl/includes/periph_bus_defines.svh <- rtl/includes/pulp_peripheral_defines.svh dependency dropped.
make: Circular rtl/includes/periph_bus_defines.svh <- rtl/includes/periph_bus_defines.svh dependency dropped.
python3 util/ioscript.py\
                                --soc-defines rtl/includes/pulp_soc_defines.svh\
                                --peripheral-defines rtl/includes/pulp_peripheral_defines.svh\
                                --periph-bus-defines rtl/includes/periph_bus_defines.svh\
                                --perdef-json perdef.json\
                                --pad-control rtl/core-v-mcu/top/pad_control.sv\
                                --xilinx-core-v-mcu-sv emulation/core-v-mcu-nexys/rtl/core_v_mcu_nexys.v\
                                --input-xdc emulation/core-v-mcu-nexys/constraints/Nexys-A7-100T-Master.xdc\
                                --output-xdc emulation/core-v-mcu-nexys/constraints/core-v-mcu-pin-assignment.xdc
Traceback (most recent call last):
  File "util/ioscript.py", line 380, in <module>
    for ionum in range(N_IO):
NameError: name 'N_IO' is not defined
make: *** [Makefile:208: rtl/core-v-mcu/top/pad_control.sv] Error 1
kyeoh1@ubuntu:/work/kyeoh1/core-v-mcu$ which ^C
kyeoh1@ubuntu:/work/kyeoh1/core-v-mcu$ python3 util/ioscript.py\
>                                 --soc-defines rtl/includes/pulp_soc_defines.svh\
>                                 --peripheral-defines rtl/includes/pulp_peripheral_defines.svh\
>                                 --periph-bus-defines rtl/includes/periph_bus_defines.svh\
>                                 --perdef-json perdef.json\
>                                 --pad-control rtl/core-v-mcu/top/pad_control.sv\
>                                 --xilinx-core-v-mcu-sv emulation/core-v-mcu-nexys/rtl/core_v_mcu_nexys.v\
>                                 --input-xdc emulation/core-v-mcu-nexys/constraints/Nexys-A7-100T-Master.xdc\
>                                 --output-xdc emulation/core-v-mcu-nexys/constraints/core-v-mcu-pin-assignment.xdc
Traceback (most recent call last):
  File "util/ioscript.py", line 380, in <module>
    for ionum in range(N_IO):
NameError: name 'N_IO' is not defined

mertkorkut, 2023-07-31:
Hi.

Could you solve the problem? I get the same error.

MikeOpenHWGroup, 2023-07-31:
Thanks for your interest in CORE-V-MCU @mertkorkut and @kyeoh2 and thanks for bringing this issue to our attention.  The `src` target is no longer supported as the MCU has dropped the ability to generate configurations of the MCU.  As of now, we support a single, static configuration.

The `src` target should be depreciated altogether.  I will be preparing a PR for this shortly.