openhwfoundation/core-v-mcu #236: Instruction needed to run cli_test in simulation

jaedenzou, 2022-07-04:
Hi, 

In the quick start guide, it is mentioned that "After working through this document you should have a [cli_test](https://github.com/openhwgroup/core-v-mcu-cli-test) running on the CORE-V-MCU on either an FPGA based emulation platform or in simulation using Verilator". The majority of the quick start guide introduced how to run it on FPGA specifically. We want to try it out in simulation first since we don't have a board you specified in hand. But I didn't find documentation on how to run cli_test in sumulation similar to the emulation where we can interact with a terminal to verify. Could you help by giving some instruction on that?  Another question by the way, does this board, Xilinx Virtex-7 FPGA VC707 Evaluation Kit, work for the cli_test emulation? We have this board in hand. Thank you.

davideschiavone, 2022-07-04:
hi @jaedenzou , the verilator simulation is work in progress (there is PR opened that needs to be fixed by myself asap).
if you clone that branch you should be able to use it.

Not sure your board will work, if it has enough resources there should not be any problem

jaedenzou, 2022-07-05:
> hi @jaedenzou , the verilator simulation is work in progress (there is PR opened that needs to be fixed by myself asap). if you clone that branch you should be able to use it.
> 
> Not sure your board will work, if it has enough resources there should not be any problem

Hi, @davideschiavone, thanks a lot for the reply and clarification.

davideschiavone, 2022-07-05:
i will keep it open till #230 is merged