openhwfoundation/core-v-mcu #238: cli_test's interrupt event 31 test

jaedenzou, 2022-08-03:
Hi, all,

I've been working on enable the emulation of core-v-mcu on an Altera FPGA board. Currently I can get to cli_test and most of the tests are either passing or I'm able to make out the reason of failed test. Only the event 31 of interrupt tests failure is a bit confusing to me. I'm wondering is it normal to fail this test? And if it's not, I might need some insights on what exactly is the event 31 test doing and what usually may be the reason behind its failure? 

What I see is in the cli_interrupt_tests.c file inside the cli_test repo, it's trying to trigger the 31 error event, by running a gpio event test without enabling interrupt 11. Could someone provide a further explanation on this? And does the GPIO needs to connect to anything externally to correctly run this test? Thank you very much in advance, it would be very helpful for my learning on this project.

gmartin102, 2022-08-03:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

I believe event 31 test does not pass.  The event31 is generated if any of the other event fifos overflow.  I was not the original designer of the event logic, but my understanding is each event has a 4 entry fifo that captures event (interrupts to my minds interpretation).  The various events are prioritized and presented to the CPU via interrupt 11.  The idea behind the test was to enable an event what wouldn’t be presented to the CPU and and see if the event fifo overflowed, causing an event31 to be recorded.    I don’t believe it was fully debugged.

Greg

jaedenzou, 2022-08-04:
Thank you again for the reply!

closed 2022-08-04 by jaedenzou