openhwfoundation/core-v-mcu #252: [TASK] <User Manual: CV32E40P Subsystem>

DBees, 2022-09-19:
### Is there an existing core-v-mcu task for this?

- [X] I have searched the existing task issues

### Task Description

Complete CV32E40P subsystem section of User Manual

### Description of Done

An accepted Pull Request to the development2 branch with the following contents: 
- CV32E40P IP Configuration
VendorID, archID, implID
- Short reprise of the processor description taken from the CV32E40P doc
- A link to the read the docs description of the processor
- Interconnections
- All the interfaces need to be mentioned

MikeOpenHWGroup, 2023-01-11:
Assigning this to myself since I have already started on it.  @davideschiavone, I will ask for a review once I've got a pull-request ready.

MikeOpenHWGroup, 2023-01-31:
Quick update: up until the end of last week (01/27) I've been focused on the CVA6 Kick-off, and have not moved this task forward.  With the kick-off, I expect to be able to devote some time to this in the coming week.