openhwfoundation/core-v-mcu #258: [TASK] <MCU manual: Logarithmic Interconnect Peripherals>

DBees, 2022-09-19:
### Is there an existing core-v-mcu task for this?

- [X] I have searched the existing task issues

### Task Description

Logarithmic Interconnect and attached peripherals section of MCU user manual

### Description of Done

Accepted pull request to the documentation2 branch regarding Theory of operation of the logarithmic interconnect and the folllowing peripherals 
Logarithmic Interconnect Peripherals

uDMA
Description of operation
DMA connected peripherals
UART
I2C
SPI Master
Camera

MikeOpenHWGroup, 2023-01-11:
Assigning this to myself as a I already made a start on it.

Hi @davideschiavone, I will ask you for a review when it is complete.

MikeOpenHWGroup, 2023-01-31:
Quick update: up until the end of last week (01/27) I've been focused on the CVA6 Kick-off, and have not moved this task forward.  With the kick-off, I expect to be able to devote some time to this in the coming week.