openhwfoundation/core-v-mcu #261: [TASK] <MCU manual: eFPGA SubSystem>

DBees, 2022-09-21:
### Is there an existing core-v-mcu task for this?

- [X] I have searched the existing task issues

### Task Description

Create a pull request covering eFPGA subsystem part of user manual on documentation2 branch

### Description of Done

Accepted pull request covering
Introduction
Sub-system Architecture 
FPGA sub-system components
Functional Description 
Logic Cell
Math Units
Block RAM 
FIFO Controller
eFPGA  Clocking
SoC Interfaces
Configurable Input/Output Signals 
SoC APB Interface
SoC TCDM Interface
SoC Interrupts

eFPGA Bitstream Loading
FPGA Configuration Control
FPGA sub-system registers

MikeOpenHWGroup, 2023-01-11:
Hello @gmartin102.  The CORE-V-MCU User Manual has undergone a significant re-organization and I would like somebody from Quicklogic to provide user-manual level documentation for the eFPGA. Can you discuss this internally and let me know who will drive this?

DBees, 2023-05-31:
https://github.com/openhwgroup/core-v-mcu/blob/master/docs/doc-src/efpga_subsystem.rst has been partially written