openhwfoundation/core-v-mcu #262: [TASK] <MCU Manual: Debug Approach>

DBees, 2022-09-21:
### Is there an existing core-v-mcu task for this?

- [X] I have searched the existing task issues

### Task Description

Create pull request covering debug approach part of user manual on documentation2 branch

### Description of Done

Accepted pull request covering debug approach

MikeOpenHWGroup, 2023-01-11:
Hi @jeremybennett, the [Debug Approach](https://docs.openhwgroup.org/projects/core-v-mcu/doc-src/debug_approach.html) chapter of the CORE-V-MCU User Manual was assigned to you a while back.  Would you or a delegate be able to put in some time to populate this chapter?

jeremybennett, 2023-01-18:
@MikeOpenHWGroup Thanks for the heads up. Will look at this.