openhwfoundation/core-v-mcu #265: [TASK] <MCU Manual: High Level Architecture>

DBees, 2022-09-21:
### Is there an existing core-v-mcu task for this?

- [X] I have searched the existing task issues

### Task Description

Create pull request on documentation2 branch covering high-level architecture 

### Description of Done

Accepted pull request covering
CORE-V MCU block diagram with main sub-systems
The top-level internal system elements and peripherals in the CORE-V MCU
Key blocks in the main sub-systems
Buses in the CORE-V MCU: APB, secondary buses, and bridges
Memory concepts in the CORE-V MCU architecture
SRAM
Private
TCDM
Logarithmic interconnect bus

MikeOpenHWGroup, 2023-01-11:
Hi @davideschiavone.  The updated CORE-V-MCU has a new chapter called [High Level Architecture](https://docs.openhwgroup.org/projects/core-v-mcu/doc-src/high_level_architecture.html).   It is "OK" and "mostly complete", but it would benefit from a review from you.  Please have a look and send me your feedback.

cst-aditya, 2025-03-18:
Hi @MikeOpenHWGroup ,
cc: @DBees , @davideschiavone 

Based on the CORE-V-MCU block diagram, I have a question. There are places in the diagram below (marked with a red circle) where the arrow's direction is not shown. Could you please assist to make this clear? 

[image]

**PS:** I refrained from opening a new issue because this one is connected to MCU Manual: High Level Architecture. I hope the call was the right one.

MikeOpenHWGroup, 2025-03-18:
Hi @cst-aditya, these shown be shown as bi-directional interfaces.  For the most part, those arrows represent the APB interface between the peripheral block and the APB Interconnect.  There are some additional signals, but these do not need to be illustrated at this level.

I've updated the diagram and can create a pull-request to update the Figure in the User Manual.  I'll wait for a few more weeks to see if there are additional questions before doing that.

cst-aditya, 2025-03-18:
Hi @MikeOpenHWGroup , Thank you for the updates. I understand.