openhwfoundation/core-v-mcu #271: Discrepancy between memory sizes within CORE-V-MCU User Manual

MikeOpenHWGroup, 2022-11-22:
### Is there an existing core-v-mcu bug for this?

- [X] I have searched the existing bug issues

### Bug Description

This issue is either related to, or affects the outcome of #265 and #45.

The Block Diagram of the MCU in the [User Manual](https://docs.openhwgroup.org/projects/core-v-mcu/doc-src/overview.html) indicates that the on-chip `Interleaved RAM` is comprized of four banks of 192Kbytes each (for a total of 768Kbytes).  However, the [memory map](https://docs.openhwgroup.org/projects/core-v-mcu/doc-src/mmap.html#memory-map) indicates that the address range is 0x1c01_0000 to 0x1c08_ffff which is 0x8_0000 or 512Kbytes.  A quick look at the RTL for both simulation and FPGA models agrees with the memory map.

So either:
1. The Block Diagram is wrong and should be updated, or
2. The ASIC implementation uses SRAMs that are indeed 192Kbytes each.

In either case, the User Manual should be updated.

@gmartin102 do you know how large the ASIC SRAMs are?

MikeOpenHWGroup, 2022-11-23:
Copying Greg's email response here:
```
Mike,

The total ram is 512 Kbytes
It is comprised of 32 16Kbyte RAMs (4Kx32 organization)
The first 32Kbytes are private RAM0 (2 ram modules) offset 0x0-0x7fff
The second 32Kytes are private RAM1 (2  ram modules) offset 0x8000-0xffff
The remaining ram is the interleaved ram.  4 banks of 112Kbytes (7 ram modules) 0ffset 0x10000-0x7fffff
Each bank has its address stride to be 0x10,  
address 0x10000 is in bank0, 0x4 is in bank 1, 0x8 is bank 2 and 0xC is in bank 3
address 0x10010 is in bank0, 0x14 is in bank 1, 0x18 is bank 2 and 0x1C is in bank 3

 So I think the 192 should be 112 and 64K of private ram.
```
MCU user manual has been updated, see #273.  Will close this issue when that PR is merged.

jeremybennett, 2022-11-24:
@MikeOpenHWGroup @gmartin102 

Interestingly, this is what I have in the Verilator model testbench. It was reporting the memory is good. I'll just rebuild and verify this is still the case. 
```c++
          size_t mb = args->maxBlock ();
          testsuite->testMem ("boot rom", 0x1a000000, 0x1a03ffff, mb, true);
          testsuite->testMem ("memory bank 0", 0x1c000000, 0x8000, mb, false);
          testsuite->testMem ("memory bank 1", 0x1c008000, 0x8000, mb, false);
          testsuite->testMem ("memory bank interleaved", 0x1c010000, 0x80000,
                              mb, false);
```
Second argument is start address, third argument is end address, fifth argument is `true` if the memory is read only.

jeremybennett, 2022-11-24:
@MikeOpenHWGroup @gmartin102 I think what this means is I agree with Greg's layout, except I have it all at offset 0x1c000000

Of course, it may be that top bits are just being lopped off, so my offset is entirely irrelevant...

jeremybennett, 2022-11-24:
I can confirm the Verilator model reports good memory as follows:
```
Testing read only memory region "boot rom": 0x1a000000 - 0x1a000027
Testing read only memory region "boot rom": 0x1a02b29d - 0x1a02b2a2
Testing read only memory region "boot rom": 0x1a03ffce - 0x1a03ffff
Testing read/writememory region "memory bank 0": 0x1c000000 - 0x1c00003b
Testing read/writememory region "memory bank 0": 0x1c005330 - 0x1c005354
Testing read/writememory region "memory bank 0": 0x1c007fc8 - 0x1c007fff
Testing read/writememory region "memory bank 1": 0x1c008000 - 0x1c008030
Testing read/writememory region "memory bank 1": 0x1c00962c - 0x1c009667
Testing read/writememory region "memory bank 1": 0x1c00ffed - 0x1c00ffff
Testing read/writememory region "memory bank interleaved": 0x1c010000 - 0x1c010022
Testing read/writememory region "memory bank interleaved": 0x1c07eace - 0x1c07eae5
Testing read/writememory region "memory bank interleaved": 0x1c08ffda - 0x1c08ffff
End: real time: 1043.085949409s, sim time: 37965450ns
Simulated cycles: 3796545, clock rate: 3.63972kHz
```

MikeOpenHWGroup, 2022-11-24:
This is different from what @gmartin102 reports: `Offset 0x10000-0x7fffff`.  Perhaps the FPGA version has more memory?

jeremybennett, 2022-11-24:
@MikeOpenHWGroup Good point - perhaps I dropped a zero. I can rebuild the model tomorrow and try a larger range. I didn't do the inverse test of "what memory is not there".

jeremybennett, 2022-11-25:
@MikeOpenHWGroup I note that memory up to 0x800000 would be 8MB of memory is that a mistake in @gmartin102 's figures above?

jeremybennett, 2022-11-25:
@MikeOpenHWGroup @gmartin102 I suspect the `0x800000` is a typo. There doesn't seem to be writable memory at the higher address!

```
Testing read only memory region "boot rom": 0x1a000000 - 0x1a000027
Testing read only memory region "boot rom": 0x1a02b29d - 0x1a02b2a2
Testing read only memory region "boot rom": 0x1a03ffce - 0x1a03ffff
Testing read/writememory region "memory bank 0": 0x1c000000 - 0x1c00003b
Testing read/writememory region "memory bank 0": 0x1c005330 - 0x1c005354
Testing read/writememory region "memory bank 0": 0x1c007fc8 - 0x1c007fff
Testing read/writememory region "memory bank 1": 0x1c008000 - 0x1c008030
Testing read/writememory region "memory bank 1": 0x1c00962c - 0x1c009667
Testing read/writememory region "memory bank 1": 0x1c00ffed - 0x1c00ffff
Testing read/writememory region "memory bank interleaved": 0x1c010000 - 0x1c010022
Testing read/writememory region "memory bank interleaved": 0x1c4ea00e - 0x1c4ea025
- memory bank interleaved, test zero: at 0x1c4ea00e: wrote 0x00, read back 0xdc
- memory bank interleaved, test zero: at 0x1c4ea00f: wrote 0x00, read back 0xba
- memory bank interleaved, test zero: at 0x1c4ea010: wrote 0x00, read back 0x1e
...
- memory bank interleaved, test random: at 0x1c7fffff: wrote 0x83, read back 0xba
End: real time: 989.693062414s, sim time: 37965450ns
Simulated cycles: 3796545, clock rate: 3.83608kHz
```

MikeOpenHWGroup, 2022-11-25:
Agreed, for the FPGA implementation, memory tops out at 0x1c08ffff.  I'm still looking into the ASIC and simulation models to see if they agree.  This will be useful information for the User Manual.

In the meantime, I'll close this issue (but will probably ask @jeremybennett and @gmartin102 to review my User Manual updates when I'm ready to merge them in).

closed 2022-11-25 by MikeOpenHWGroup