openhwfoundation/core-v-mcu #274: [BUG] couldn't find in the docs or QSG which Vivado version is needed

mbaykenar, 2022-11-24:
### Is there an existing core-v-mcu bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hi, which version of Vivado is supported? In the code I searched for vivado:

rtl/vendor/pulp_platform_riscv_dbg/src/dm_csrs.sv --> // Xilinx Vivado 2020.1 does not allow subtraction of two enums; do the subtraction with logic
rtl/vendor/pulp_platform_axi/CHANGELOG.md --> ## 0.9.1 - 2020-01-18

different vivado versions sometimes creates compatibility issues due to XILINX IP revisions. I looked at IPs at emulation/xilinx/ips and only 

- clk_wiz
- blk_mem_gen

seems to be used for generating bitstream with Vivado. Is it right?

It could be great if repo has some info about Vivado version, or if exists easier to find

MikeOpenHWGroup, 2022-11-28:
Thanks for the feedback @mbaykenar, you are right that we should include this!

I have successfully built a bitmap for the Nexys A7 using Vivado 2021.1 (64-bit) using the following commands:
```
$ git clone https://github.com/openhwgroup/core-v-mcu
$ cd core-v-mcu make nexys-emul
```

MikeOpenHWGroup, 2022-11-29:
Hi again @mbaykenar.  I think (hope) this meets your needs.  Please close the issue if it does.  Thanks.

mbaykenar, 2022-11-29:
great! Thanks a lot!

closed 2022-11-29 by mbaykenar