openhwfoundation/core-v-mcu #275: [BUG] pip3 install --user -r python-requirements.txt gives error

mbaykenar, 2022-11-24:
### Is there an existing core-v-mcu bug for this?

- [X] I have searched the existing bug issues

### Bug Description

After cloning the repo I simply run 

pip3 install --user -r python-requirements.txt

the run gives an error:

ERROR: importlib-resources 5.10.0 has requirement zipp>=3.1.0; python_version < "3.10", but you'll have zipp 1.0.0 which is incompatible.

MikeOpenHWGroup, 2022-11-28:
Thanks for the report @mbaykenar.  I have not run this in a while and I too see this error.  It seems that the problem is with `yamlfmt`.  [This](https://stackoverflow.com/questions/49868091/python-error-when-installing-ruamel-yaml-package-with-pip) post in stackoverflow suggests down-versioning from python3.10 to 3.9.  This does not seem to be an intelligent solution.

Continuing to investgate...

mbaykenar, 2022-12-05:
Hi @MikeOpenHWGroup,

I confirm that downgrading python to 3.9 solved the issue. Thanks a lot.

closed 2022-12-05 by mbaykenar