openhwfoundation/core-v-mcu #277: [BUG] Quickstart guide - replace Eclipse IDE with SDK

DBees, 2022-11-29:
### Is there an existing core-v-mcu bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Update Quickstart Guide with references to SDK

MikeOpenHWGroup, 2022-11-29:
Do you mean [this](https://github.com/openhwgroup/core-v-mcu/tree/master/emulation/quickstart#eclipse-ide) section?  What SDK should we specify?

DBees, 2022-11-29:
Yes, that is the section. I believe we should specify https://github.com/openhwgroup/core-v-sdk

MikeOpenHWGroup, 2022-11-29:
Fair enough.  Is this urgent?  I have not yet used the SDK and it would take me several days to figure it all out before I could write it up for the QSG.  Is there another member I can ask?

DBees, 2022-11-30:
It is not urgent, but important. I assigned this to myself