openhwfoundation/core-v-mcu #286: Add programming example for micro-DMA function

MikeOpenHWGroup, 2023-01-12:
### Is there an existing core-v-mcu task for this?

- [X] I have searched the existing task issues

### Task Description

Currently the micro-DMA chapter of the User Manual is a description of how the uDMA works.  We need a programmer's view that explains how to write/read the uDMA and peripheral CSRs to transfer data to/from the L2 memories.

Let's start with a simple peripheral for this.  I recommend either the uDMA UART or uDMA I2CM peripherals.

### Description of Done

Merge pull-request attributed to this isse.

DBees, 2023-01-12:
@MikeOpenHWGroup are you advocating a new chapter or augmenting the existing micro-DMA chapter?

MikeOpenHWGroup, 2023-01-12:
> @MikeOpenHWGroup are you advocating a new chapter or augmenting the existing micro-DMA chapter?

I would recommend adding this to the peripheral function chapter (e.g. uDMA UART).

abberthe, 2023-01-12:
> > @MikeOpenHWGroup are you advocating a new chapter or augmenting the existing micro-DMA chapter?
> 
> I would recommend adding this to the peripheral function chapter (e.g. uDMA UART).

Under the Programming Model Section?

abberthe, 2023-02-09:
I started working on this. The programming sequence for all uDMA peripheral follow roughly  the same scheme, only Step 1 and Step 2 make use of the general uDMA configuration registers. The other steps involve the specific register section of the each peripheral inside the uDMA memory map.

1. Clear the clock gating bit of the associated peripheral
2. Enable interrupt on the associated peripheral
3. Setup the peripheral
4. Setup the transfer parameters (address and size)
5. Start operation (tx/rx)

We could add this generic description to the uDMA section and add the specifics for each uDMA peripheral. @MikeOpenHWGroup what do you say?

MikeOpenHWGroup, 2023-02-10:
That works for me @abberthe.  It will be important to specify the specific CSRs and CSR fields that s/w needs to write/read for each the above steps.  Of course, step 3 will be peripheral dependent, so we could simply point the reader to the appropriate section of the User Manual (e.g. uDMA UART).