openhwfoundation/core-v-mcu #287: Complete the APB Advance Timer chapter of the User Manual

MikeOpenHWGroup, 2023-01-13:
### Is there an existing core-v-mcu bug for this?

- [X] I have searched the existing bug issues

### Bug Description

A (mostly empty) shell of the APB Advanced Timer peripheral exists at `docs/doc-src/ip-blocks/apb_adv_timer.rst`.  This chapter needs to be updated such that the Reader understands _what_ the peripheral does, a high-level description of _how_ it works, and a description of how software should use the peripheral's CSRs to control its operation.

You are free to edit the chapter as you see fit, but a template has been provided with the following structure:

- **Features**: short description (can be a bullet list) of what the peripheral does.
- **Theory of Operation**: high-level description of the implementation (RTL).  Do _not_ go into too much detail, this is not a micro-architecture specification.
- **Programming Model**: this section should answer the question: what does software need to do in order to configure/control/monitor the peripheral's function.
- **CSRs**: one table per CSR.  An example is provided.

DBees, 2023-01-16:
@MikeOpenHWGroup The task is clearly laid out, thank you. Is the assignee expected to reverse engineer the code or is there any documentation we can find from ETH?

MikeOpenHWGroup, 2023-01-17:
That is a good point @DBees.  I'll add something about that in "background information".