openhwfoundation/core-v-mcu #294: portmacro.h file error and creating my own app using c compiler

WazaAbdulkadir, 2023-02-10:
I want to implement my own application. I have created a source code with necessary header files to blink led0. 

I need to compile for rv32 architecture. For this i think using the necessary portmacro.h file will solve my problem.

I have implemented both of portmacro header files available in core-v-mcu cli-test repository:
1- [link1](https://github.com/openhwgroup/core-v-mcu-cli-test/blob/3358348f8bcdb5acffa9a3233cc05301d2acd618/cli_test/kernel/portable/GCC/RISC-V/portmacro.h) 
2-[link2]( https://github.com/openhwgroup/core-v-mcu-cli-test/blob/3358348f8bcdb5acffa9a3233cc05301d2acd618/cli_test/kernel/portable/IAR/RISC-V/portmacro.h )

For the first link it gave this error: #error Assembler did not define __riscv_xlen 

For second one I couldn't find the necessity "intrinsics.h" file. 

I think I need to run the c code for riscv architecture, but when i run it it automaticaly runs it for x86 arc.

I am planning to create a bin file and load it using serialPort.py. Is this possible ?

jeremybennett, 2023-02-10:
Hi @WazaAbdulkadir 

Not sure where you are getting your toolchains.  You can download pre-compiled RISC-V and CORE-V tool chains [here](https://www.embecosm.com/resources/tool-chain-downloads/).  Just checked the macros with the latest CORE-V tool chain and `__riscv_xlen` is there.  Here is how you dump out all the macros with "riscv" in the name from GCC.
```
$ riscv32-corev-elf-gcc -march=rv32imac -mabi=ilp32 -E -dM - < /dev/null | grep -i riscv
#define __riscv 1
#define __riscv_atomic 1
#define __riscv_float_abi_soft 1
#define __riscv_cmodel_medlow 1
#define __riscv_mul 1
#define __riscv_muldiv 1
#define __riscv_xlen 32
#define __riscv_m 2000000
#define __riscv_a 2001000
#define __riscv_c 2000000
#define __riscv_i 2001000
#define __riscv_compressed 1
#define __riscv_arch_test 1
#define __riscv_div 1
```

HTH

MikeOpenHWGroup, 2023-02-10:
Hi @WazaAbdulkadir, thanks for your issue.  Please respond to this issue to confirm (or not) that the advice from @jeremybennett worked for you.  If so, please close this issue.

There are a couple of other resources that you may or may not be aware of:
- [core-v-mcu-cli-test](https://github.com/openhwgroup/core-v-mcu-cli-test) is a test-program environment for the CORE-V-MCU running on freeRTOS.
- The User Manual for the CORE-V-MCU on ReadTheDocs ([link](https://docs.openhwgroup.org/projects/core-v-mcu/doc-src/high_level_architecture.html)).

You will note that the `cli-test` is complete, at least in the sense that it provides a framework to exercise the MCU.  The [Quick Start Guide](https://github.com/openhwgroup/core-v-mcu/blob/master/emulation/quickstart/README.md) will get you blinking an LED in a few minutes. We are always open to new contributions.  The MCU User Manual is less mature and in active development.  Questions about, and/or contributions to, the User Manual most welcome.

closed 2023-02-13 by WazaAbdulkadir