openhwfoundation/core-v-mcu #296: [BUG] <title>

mbaykenar, 2023-02-21:
### Is there an existing core-v-mcu bug for this?

- [X] I have searched the existing bug issues

### Bug Description

In apb_soc_ctrl.sv some definitions for register offsets are:

`define REG_WCFGFUN 12'h60 // BASEADDR+0x60 Sets mux&cfg control for specifed iopad
`define REG_RCFGFUN 12'h64 // BASEADDR+0x64 reads mux&cfg control for specifed iopad

`define REG_JTAGREG 12'h74 //BASEADDR+0x74 JTAG REG
`define REG_BOOTSEL 12'hC4 //BASEADDR+0xC4 bootsel
`define REG_CLKSEL 12'hC8 //BASEADDR+0xC8 clocksel
`define REG_WD_COUNT 12'hD0
`define REG_WD_CONTROL 12'hD4
`define REG_RESET_REASON 12'hD8
`define RTO_PERIPHERAL 12'hE0
`define RTO_COUNT 12'hE4
`define RESET_TYPE1_EFPGA 12'hE8 //BASEADDR+0xE8
`define ENABLE_IN_OUT_EFPGA 12'hEC //BASEADDR+0xEC
`define EFPGA_CONTROL 12'hF0
`define EFPGA_STATUS 12'hF4
`define EFPGA_VERSION 12'hF8
`define SOFT_RESET 12'hFC

`define PAD_CFG_MUX 12'b0100????????  // 0x400 - 7FC for 256 PADMUX

I was reading the documentation for apb_soc_ctrl section and IO_CTRL offset is seem to be 0x0400, which is PAD_CFG_MUX parameter in sv file. However, this parameter is never used in this file or another in the repo.

This offset val is also seen in hal_apb_soc_ctrl_reg_defs.h in hal library for core-v-mcu:

  // Offset = 0x0400
  union {
    __IO uint32_t io_ctrl[48];
    struct {
      __IO uint32_t  mux        :  2;
      __IO uint32_t             :  6;
      __IO uint32_t  cfg        :  6;
    } io_ctrl_b[48];
  };
} SocCtrl_t;

Correct offset is seem to be REG_WCFGFUN ?

or I simply did not understand the meaning of 0x400 offset for I/O control ?

Regards,

MikeOpenHWGroup, 2023-02-21:
Hi @mbaykenar, Thanks for your interest in CORE-V-MCU.  I took the liberty of editing your issue description to clean up the markdown formatting.   You have asked a couple of questions which I will try to answer.  These answers reference the [CORE-V-MCU User's Manual](https://docs.openhwgroup.org/projects/core-v-mcu/index.html), which is currently under development.

1. You are right, the `PAD_CFG_MUX` macro is not used.  It will be deleted from a future release of the RTL.
2. Addressing for CORE-V-MCU is determined by adding a base address an offset.  The base addresses are in the [memory map](https://docs.openhwgroup.org/projects/core-v-mcu/doc-src/mmap.html), and the offset is provided in the documentation for each IP Block.

In the case of the [APB SoC Event Controller](https://docs.openhwgroup.org/projects/core-v-mcu/doc-src/ip-blocks/apb_soc_ctrl.html), the base address is `0x1A10-6000` and the offset for `IO_CFG0` is 0x0400, so the address of the `IO_CFG0` CSR in the APB SoC Event Controller is 0x1A10-6400.

You will note that the documentation provides information about a set of CSRs called `IO_CTRL` and you will not find these in the RTL.  Also, the RTL has an address `REG_WCFGFUN` and you will not find this in the documentation.  That is because the CORE-V-MCU User Manual is a *user's* document, not an RTL *designers* document.

Lastly, we are in the process of creating a significant update to the CORE-V-MCU User Manual, and your questions are most welcome.  If you have the time and interest, perhaps you would be willing to help out - let me know!

MikeOpenHWGroup, 2023-02-21:
I hope this answers your question @mbaykenar.  If it does, please close this issue.  If you have another question, feel free to create a new issue. We try hard not to resolve multiple bugs/questions in a single issue.

Thanks!

mbaykenar, 2023-02-21:
@MikeOpenHWGroup thanks a lot for your time and detailed answers. I understand that the documentation is an ongoing process, I am working on core-v-mcu nowadays to understand and modify the architecture for a commercial design.

Just a note, I don't think my question is related to "APB SoC Event Controller" but rather related to "APB SoC Controller", whose base address is 0x1A104000. But anyways I got my answer, thanks a lot.

Regarding to the contribution or helping core-v-mcu, I will be honoured to be in such a position. I will sent a personal email to you

Regards,

closed 2023-02-21 by mbaykenar