openhwfoundation/core-v-mcu #301: [BUG]<build verilator model library>

chanduputta, 2023-03-01:
### Is there an existing core-v-mcu bug for this?

- [X] I have searched the existing bug issues

### Bug Description

To do simulation using verilator and gtkwave
we are trying to "build verilator model library"

> System details

Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.1 LTS
Release:	22.04
Codename:	jammy

by running command `$ make model-lib`

> fusesoc --cores-root . run --target=model-lib --setup \
	--build openhwgroup.org:systems:core-v-mcu | tee model-lib.log
INFO: Preparing openhwgroup.org:ip:apb2apbcomp:0
INFO: Preparing openhwgroup.org:ip:apb2per:0
INFO: Preparing openhwgroup.org:ip:apb_adv_timer:0
INFO: Preparing openhwgroup.org:ip:apb_fll_if:0
INFO: Preparing openhwgroup.org:ip:apb_gpio:0
INFO: Preparing openhwgroup.org:ip:apb_i2cs:0
INFO: Preparing openhwgroup.org:ip:apb_node:0
INFO: Preparing openhwgroup.org:ip:apb_timer_unit:0
INFO: Preparing openhwgroup.org:ip:axi_slice:0
INFO: Preparing openhwgroup.org:ip:cv32e40p:0
INFO: Preparing openhwgroup.org:ip:logint_dc_fifo_xbar:0
INFO: Preparing openhwgroup.org:ip:tcdm_interconnect:0
INFO: Preparing openhwgroup.org:ip:udma_camera:0
INFO: Preparing openhwgroup.org:ip:udma_core:0
INFO: Preparing openhwgroup.org:ip:udma_external_per:0
INFO: Preparing openhwgroup.org:ip:udma_filter:0
INFO: Preparing openhwgroup.org:ip:udma_i2c:0
INFO: Preparing openhwgroup.org:ip:udma_i2s:0
INFO: Preparing openhwgroup.org:ip:udma_qspi:0
INFO: Preparing openhwgroup.org:ip:udma_sdio:0
INFO: Preparing openhwgroup.org:ip:udma_uart:0
INFO: Preparing pulp-platform.org::common_cells:1.20.0
INFO: Preparing pulp-platform.org::tech_cells_generic:0
INFO: Preparing quicklogic.com:ip:efpga:0
INFO: Preparing openhwgroup.org:ip:l2_tcdm_hybrid_interco:0
INFO: Preparing pulp-platform.org::axi:0.28.0
INFO: Preparing pulp-platform.org::fpnew:0
INFO: Preparing pulp-platform.org::fpu_div_sqrt_mvp:0
INFO: Preparing pulp-platform.org::riscv_dbg:0
INFO: Preparing openhwgroup.org:systems:core-v-mcu:0
INFO: Setting up project
INFO: Running pre_build script pre_build_scripts
sh: 0: cannot open ../src/openhwgroup.org_systems_core-v-mcu_0/rtl/core-v-mcu/scripts/vedit.sh: No such file
ERROR: Failed to build openhwgroup.org:systems:core-v-mcu:0 : pre_build script 'pre_build_scripts': ['sh', '../src/openhwgroup.org_systems_core-v-mcu_0/rtl/core-v-mcu/scripts/vedit.sh', 'openhwgroup.org_systems_core-v-mcu_0.vc'] exited with error code 2

ERROR: Failed to build openhwgroup.org:systems:core-v-mcu:0 : pre_build script 'pre_build_scripts': ['sh', '../src/openhwgroup.org_systems_core-v-mcu_0/rtl/core-v-mcu/scripts/vedit.sh', 'openhwgroup.org_systems_core-v-mcu_0.vc'] exited with error code 2

Do we need to setup system with some other tool-chains or packages for simulation? any advice will be appreciated?

jeremybennett, 2023-03-01:
Hi @chanduputta 

I have just tried the latest source (commit 75f8df9) without problems. I was running on Ubuntu 22.04 using Verilator 4.203. Can you advise of the operating system and Verilator version you are using? The failed script is very simple, just invoking _sed_.  Asuming it was _sed_ which failed, the erro code 2, means:

>One or more of the input file specified on the command line could not be opened (e.g.  if a file is not found, or read permission is denied).  Processing continued with other files.

The script only attempts to open two files, `openhwgroup.org_systems_core-v-mcu_0.vc`, and `Makefile` which are generated in the build directory.

I've assigned to @MikeOpenHWGroup in case he has more insight.

chanduputta, 2023-03-01:
hi @jeremybennett Thank you responce
the Verilator version and operating system we using are:

> verilator --version
Verilator 5.003 devel rev v5.002-86-ge97e6c221

> Operating System: Ubuntu 22.04.1 LTS              
          Kernel: Linux 5.14.0-1057-oem
    Architecture: x86-64
 Hardware Vendor: Dell Inc.
  Hardware Model: Latitude 3520

jeremybennett, 2023-03-01:
@chanduputta Well you have me stumped. That's not dissimilar to my platform.  The only thing I can suggest is hacking `vedit.sh` (it's only a few lines long) to get it to print out diagnostics so you can find out what file it can't see. Maybe even just add `set -x` so it prints out each command as it executes it.

MikeOpenHWGroup, 2023-03-01:
I note that @jeremybennett is using Verilator v4.203 and @chanduputta is using v5.003.  I have seen this behavior many times with the CVA6: you have to pick the just right version of Verilator to get the DUT and/or testbench to compile.  Very annoying.

@chanduputta can you try v4.203?

chanduputta, 2023-03-02:
Hi @MikeOpenHWGroup we are not able to find Verilator v4.203 even in there github tags(ref below screenshot)

we have tried in another OS and verilator version 

>  Operating System: Ubuntu 20.04.5 LTS
            Kernel: Linux 5.15.0-60-generic
      Architecture: x86-64

> $ verilator --version
Verilator 4.028 2020-02-06 rev v4.026-92-g890cecc1

by running command `$ make model-lib`  we got again an error at Building simulation model
> $ sudo make model-lib
fusesoc --cores-root . run --target=model-lib --setup \
	--build openhwgroup.org:systems:core-v-mcu | tee model-lib.log
INFO: Preparing openhwgroup.org:ip:apb2apbcomp:0
INFO: Preparing openhwgroup.org:ip:apb2per:0
INFO: Preparing openhwgroup.org:ip:apb_adv_timer:0
INFO: Preparing openhwgroup.org:ip:apb_fll_if:0
INFO: Preparing openhwgroup.org:ip:apb_gpio:0
INFO: Preparing openhwgroup.org:ip:apb_i2cs:0
INFO: Preparing openhwgroup.org:ip:apb_node:0
INFO: Preparing openhwgroup.org:ip:apb_timer_unit:0
INFO: Preparing openhwgroup.org:ip:axi_slice:0
INFO: Preparing openhwgroup.org:ip:cv32e40p:0
INFO: Preparing openhwgroup.org:ip:logint_dc_fifo_xbar:0
INFO: Preparing openhwgroup.org:ip:tcdm_interconnect:0
INFO: Preparing openhwgroup.org:ip:udma_camera:0
INFO: Preparing openhwgroup.org:ip:udma_core:0
INFO: Preparing openhwgroup.org:ip:udma_external_per:0
INFO: Preparing openhwgroup.org:ip:udma_filter:0
INFO: Preparing openhwgroup.org:ip:udma_i2c:0
INFO: Preparing openhwgroup.org:ip:udma_i2s:0
INFO: Preparing openhwgroup.org:ip:udma_qspi:0
INFO: Preparing openhwgroup.org:ip:udma_sdio:0
INFO: Preparing openhwgroup.org:ip:udma_uart:0
INFO: Preparing pulp-platform.org::common_cells:1.20.0
INFO: Preparing pulp-platform.org::tech_cells_generic:0
INFO: Preparing quicklogic.com:ip:efpga:0
INFO: Preparing openhwgroup.org:ip:l2_tcdm_hybrid_interco:0
INFO: Preparing pulp-platform.org::axi:0.28.0
INFO: Preparing pulp-platform.org::fpnew:0
INFO: Preparing pulp-platform.org::fpu_div_sqrt_mvp:0
INFO: Preparing pulp-platform.org::riscv_dbg:0
INFO: Preparing openhwgroup.org:systems:core-v-mcu:0
INFO: Setting up project
INFO: Running pre_build script pre_build_scripts
INFO: Building simulation model
make[1]: Entering directory '/home/shreyashbantu/my-project/core-v-mcu/build/openhwgroup.org_systems_core-v-mcu_0/model-lib-verilator'
verilator -f openhwgroup.org_systems_core-v-mcu_0.vc -Wno-fatal -O3 --trace --CFLAGS -fPIC -DVL_TIME_CONTEXT
%Error: ../src/pulp-platform.org__tech_cells_generic_0/pulp_platform_tech_cells_generic/src/rtl/tc_clk.sv:80: syntax error, unexpected TIME NUMBER, expecting TYPE-IDENTIFIER
  parameter int unsigned Delay = 300ps
                                 ^~~~~
%Error: ../src/pulp-platform.org__tech_cells_generic_0/pulp_platform_tech_cells_generic/src/rtl/tc_clk.sv:83: syntax error, unexpected output, expecting IDENTIFIER or '=' or do or final
  output logic out_o
  ^~~~~~
%Error: ../src/pulp-platform.org__tech_cells_generic_0/pulp_platform_tech_cells_generic/src/rtl/tc_sram.sv:93: Unsupported or unknown PLI call: $urandom
          "random": init_val[i][j] = $urandom();
                                     ^~~~~~~~
%Warning-STMTDLY: ../src/openhwgroup.org_systems_core-v-mcu_0/rtl/core-v-mcu/components/freq_meter.sv:33: Unsupported: Ignoring delay on this delayed statement.
    #10 rstn = 1;
    ^
                  ... Use "/* verilator lint_off STMTDLY */" and lint_on around source to disable this message.
%Error: Exiting due to 3 error(s)
make[1]: *** [Makefile:16: Vcore_v_mcu.mk] Error 1
make[1]: Leaving directory '/home/shreyashbantu/my-project/core-v-mcu/build/openhwgroup.org_systems_core-v-mcu_0/model-lib-verilator'
ERROR: Failed to build openhwgroup.org:systems:core-v-mcu:0 : '['make']' exited with an error: 2

ERROR: Failed to build openhwgroup.org:systems:core-v-mcu:0 : '['make']' exited with an error: 2

reference:
[image]

MikeOpenHWGroup, 2023-03-02:
I am able to compile with Verilator v4.100.  I suggest you built it from source.  Here is what I do:
```
$ cd my_working_dir
$ git clone https://github.com/verilator/verilator
$ cd verilator
$ git tag                  # See what versions exist
$ git checkout v{version}  # Switch to specified release version
$ autoconf                 # Create ./configure script
$ ./configure
$ export VERILATOR_ROOT="my_working_dir/verilator/v{version}"
```

Also, make sure `rtl/includes/pulp_peripheral_defines.svh` has not been modified.