openhwfoundation/core-v-mcu #302: Using the A2 bootloader to program flash memory

abdullahyildiz, 2023-03-09:
It is mentioned in [CORE-V-MCU Quick Start Guide](https://github.com/openhwgroup/core-v-mcu/tree/master/emulation/quickstart#using-the-bootloader) that A2 bootloader is capable of programming the flash memory. I checked the default [A2 bootloader source code](https://github.com/openhwgroup/core-v-mcu-cli-test/tree/main/A2_boot) and couldn't have found any code block that is used for programming the flash memory.

Is it required to implement such functionality in the boot code?

MikeOpenHWGroup, 2023-03-09:
Hi @abdullahyildiz, I presume you refering to the [Using the-Bootloader](https://github.com/openhwgroup/core-v-mcu/tree/master/emulation/quickstart#using-the-bootloader) section of the Quick Start Guide, which says:
```
The A2 bootloader is capable of collecting an executable file in Srecord format via UART0 and storing it into flash.
```
I believe you are right, and that statement is incorrect.  I _believe_ (do not know for sure) that programming the flash is a feature of `cli_test`.   I will confirm this with @gmartin102 and update this issue (and create a PR to fix the QSG).

gmartin102, 2023-03-09:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Mike,
You are correct, The A2 bootloader can load and Srec file into ram and jump to the starting address.  As you mentioned the cli_test program can load an image from the uart and program the flash.

Greg

MikeOpenHWGroup, 2023-03-09:
Thanks @gmartin102.  @abdullahyildiz, please have a look at pull-request #303.

closed 2023-03-09 by abdullahyildiz

MikeOpenHWGroup, 2023-07-05:
Fixed in #312.