openhwfoundation/core-v-mcu #306: cli_test on Genesys2

yelabWang, 2023-05-13:
### Is there an existing core-v-mcu bug for this?

- [X] I have searched the existing bug issues

### Bug Description

I attempted to run the application on the Genesys2 board in the quickstart folder, but the command line was unable to return to clitest. May I ask if clitest currently supports Genesys2, and how can I verify the program on Genesys2?

[image]

[image]

MikeOpenHWGroup, 2023-05-14:
Hi @yelabWang, thanks for your interest in the CORE-V-MCU.  The Quick Start Guide does not support the Genesys2 board (and I have not personally tried it).  Having said, that the Makefiles _do_ support Genesys2, but you will need to dig through the Makefiles and scripts to built and compile it all properly.

@gmartin102 are you aware of any specific documentation provided for the Genesys2 hardware?

gmartin102, 2023-05-14:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

There is no specific documentation, but it  might be related to the fact that the genesys2 board runs at different clock rates from the than the Nexsys board.  The CLI test code will need to be modified for the correct baud rate.  How was the CLI-test program loaded into memory?  Eclipse through the Ashling JTAG or a digilent HS2?

Greg

yelabWang, 2023-05-16:
I tried using the uart burning program, but after modifying the baud rate and clock cycle, I still couldn't run clitest correctly