openhwfoundation/core-v-mcu #322: [BUG] FlashID is read incorrectly in a2_boot/main.c

janhoogerbrugge, 2023-08-29:
### Is there an existing core-v-mcu bug for this?

- [X] I have searched the existing bug issues

### Bug Description

I see:

    udma_flash_readid(lFlashID);

I assume that it should be:

    udma_flash_readid((uint32_t *) &lFlashID);

MikeOpenHWGroup, 2023-08-29:
Hi @janhoogerbrugge, thanks for your interest in CORE-V-MCU.  What file is this code in?  Is it part of _this_ repo or the [CORE-V-MCU-CLI-TEST repo](https://github.com/openhwgroup/core-v-mcu-cli-test)?

janhoogerbrugge, 2023-08-29:
You can find it here: https://github.com/openhwgroup/core-v-mcu/blob/master/a2_boot/main.c

MikeOpenHWGroup, 2023-08-30:
Hi @janhoogerbrugge, yes you are correct this is a bug and your fix is valid.  As it turns out, you've highlighted a deeper issue in `a2_boot` in which it _needs_ a flash to be present.  This suites our application, but not the general case where a flash may not be present.  We are investigating a proper fix.