openhwfoundation/core-v-mcu #327: [BUG] docs sphinx_markdown_tables __init__() takes 2 but 3 given

emelcher, 2023-11-10:
### Is there an existing core-v-mcu bug for this?

- [X] I have searched the existing bug issues

### Bug Description

make docs
...
reading sources... [  3%] doc-src/apb_peripheral_interconnect                                                                                                                                              
Extension error (sphinx_markdown_tables):
Handler <function process_tables at 0x7ffaf3765730> for event 'source-read' threw an exception (exception: __init__() takes 2 positional arguments but 3 were given)

MikeOpenHWGroup, 2023-11-10:
Hi @emelcher, thanks for your Issue.  I will investigate.  In the meantime, the docs are on-line [here](https://docs.openhwgroup.org/projects/core-v-mcu/).

MikeOpenHWGroup, 2023-11-10:
Hi @emelcher, I am unable to reproduce your problem.  For me the `make docs` command runs as expected.  Have you installed all the required [python tools](https://github.com/openhwgroup/core-v-mcu#getting-started)?  Depending on your installation set-up, you may require additional plugs-ins to be installed:
```
$ pip install pdflatex
$ apt-get install texlive-latex-base
$ apt-get install texlive-latex-extra
```
Please give this a try and let me know if it resolves your issue, and if it does, you can close this issue.  Otherwise, the issue will be closed within a from today.   Thanks!

MikeOpenHWGroup, 2023-11-13:
You'll need to resolve this on your own @emelcher - we do not have the resources to support all user platforms.

If you are able to resolve the issue for CentOS, we would be happen to receive a pull-request from you to update our scripts.

In the meantime, you can forgo the generation of a PDF on your local system.  It seems that the HTML is being generated properly.  This is the preferred option in any case - Sphinx goes a much better job of HTML rendering than it does for the PDFs.

Teimir, 2026-09-28:
_In the meantime, you can forgo the generation of a PDF on your local system. It seems that the HTML is being generated properly._ 
**It isn't**

reading sources... [  2%] doc-src/apb_peripheral_interconnect                  
Extension error (sphinx_markdown_tables):
Handler <function process_tables at 0x2b82101c7510> for event 'source-read' threw an exception (exception: __init__() takes 2 positional arguments but 3 were given)
make[2]: *** [html] Error 2

MikeOpenHWGroup, 2026-09-28:
Same response @Teimir, we do not support CentOS.

Teimir, 2026-09-28:
> Same response [@Teimir](https://github.com/Teimir), we do not support CentOS.

Ubuntu 26 also not supported? 

$ pip install pdflatex
$ apt-get install texlive-latex-base
$ apt-get install texlive-latex-extra 
^ done

[image]

MikeOpenHWGroup, 2026-09-28:
@emelcher and @Teimir: I have created an entirely new python3 virtual environment on my local Ubuntu 24.04 system and followed the directions on the top-level README and [docs/README](https://github.com/openhwfoundation/core-v-mcu/blob/master/docs/README.md).  There was a gap in docs/README (not related to this problem) that I just fixed.  Following these directions, the HTML build works for me.