openhwfoundation/core-v-mcu #329: Using PMOD SF3 on Genesys2

socmonster, 2024-05-23:
### Is there an existing core-v-mcu bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello there,

I've been working on core-v-mcu for a little bit of time. I'm currently running cli_test on Genesys2 but I've did some work on Nexys as well. The problem between the two platform is that I cannot use PMOD SF3 on Genesys2. I can work on everything except of qspi flash commands. I've checked the JXADC pin connections, and they seem OK. I also tried to use another PMOD for flash connection, but the issue is still there. I don't think this is related to software stack as well. Is there anyone worked on Genesys2 and was able to boot from flash? I guess I'm in need of to be pointed into right direction. My clock speed configurations are like below.

SYS_CLK = 20 MHz
PER_CLK = 5MHz

Kind regards,

socmonster

gmartin102, 2024-05-24:
@Socmonster I think the default voltage levels on the JXADC pins is incorrect for use with a digital PMOD.  JP6 on the genesys2 board selects the Voltage to be used on the JXDC PMOD.  it should be jumpering pins 5 and 6 for 3.3v

socmonster, 2024-05-24:
Hi @gmartin102. Yes, I think the issue might be related to this case. I'm trying to debug it. When I change the JP6 jumper as you mentioned my HS2 connection fails. When JP6 jumper is on 1 and 2 for 1.8V, HS2 is working. I'll let you know if I can manage to solve the problem.

MikeOpenHWGroup, 2025-07-28:
Hi @socmonster, this issue has gone stale.  Has it been resolved?  If so, please close it, otherwise we will close it on or after Monday, August 4, 2025.  Thanks!