openhwfoundation/core-v-mcu #330: Question: Which version of OpenOCD should be used with Core-V-MCU and CV32E40P

CLappin, 2024-06-11:
Hi all,

Which version of openocd should be used with the [pulp debugger](https://github.com/pulp-platform/riscv-dbg).

I've tried the latest version which doesn't connect correctly, I've also tried the 0.10.0 version that I had which seemed to all connection via telnet. 

Any help would be great ,
Thanks,
Ciaran

MikeOpenHWGroup, 2024-06-11:
I have 0.12.0 installed on my Ubuntu machine.   The only additional support we can offer is what is written in the [Quick Start Guide](https://github.com/openhwgroup/core-v-mcu/blob/master/emulation/quickstart/README.md#core-v-mcu-quick-start-guide).

CLappin, 2024-06-12:
Thank you @MikeOpenHWGroup  and  @gmartin102 . 
I'll take a look at the Quick Start .

closed 2024-06-12 by CLappin